pvlib/pvlib-python #1249: Allow user to set tol and maxiter for singlediode newton method

toddkarin, 2021-06-17:
The first few lines of `pvlib.singlediode` set `tol` and `maxiter` for all the solvers using the newton method:

```
from scipy.optimize import brentq, newton
from scipy.special import lambertw

# set keyword arguments for all uses of newton in this module
newton = partial(newton, tol=1e-6, maxiter=100, fprime2=None)
```

However, I would like to change `tol` and `maxiter` for my application. It would be great if these could be added instead as keyword arguments to the various functions so they can be adjusted by the user. Using a variety of singlediode model params, I have found that by setting tol=0.1 and maxiter=10, I can realize a 1.4x speedup in the `singeldiode.bishop88_mpp` algorithm while incurring a maximum error of 0.007038% and a mean absolute error of  0.000042% in calculated V_mp.

cwhanse, 2021-06-17:
At what level would they be exposed?  At the `pvlib.singlediode.bishop88_xxx` functions or at a higher level, e.g., `pvlib.pvsystem.singlediode`?

toddkarin, 2021-06-17:
At the level of bishop88_xxx would be good enough for my purposes.

cwhanse, 2021-06-18:
I support this as long as the interface is common for both the `newton` and `brentq` options.

PR #1764 (2023-06-23): Allow users to interact with root finders parameters
 - [x] Closes #1249
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
~- [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Tests added
 - [x] New code is fully documented. Includes [numpydoc](https://numpydoc.readthedocs.io/en/latest/format.html) compliant docstrings, examples, and comments where necessary.
 - [x] Pull request is nearly complete and ready for detai [... cut]
closed 2023-06-23 by kandersolar