pvlib/pvlib-python #1252: Missing Module Dependency - Tables

zhammond147, 2021-07-02:
In the pvlib.clearsky.lookup_linke_turbidity() function, you have the following error handling:

try:
    import tables
except ImportError:
    raise ImportError('The Linke turbidity lookup table requires tables. '
                                'You can still use clearsky.ineichen if you '
                                'supply your own turbidities.')

Unfortunately, the _tables_ module has not been included as a dependency of the pvlib package.

kanderso-nrel, 2021-07-02:
It is an optional dependency: https://github.com/pvlib/pvlib-python/blob/master/setup.py#L56
See also https://pvlib-python.readthedocs.io/en/stable/installation.html#compatibility

Should this be explained better in the online docs?  Or maybe mention something about `pvlib[optional]` in that error message?

cwhanse, 2021-07-02:
Is there a good reason that `tables` is still optional? This is a frequent pothole hit by users.

mikofski, 2021-08-27:
BTW: I though TL data was originally available for download as an `.nc` file, am I wrong? I guess it's not available from [soda pro](http://www.soda-pro.com/help/general-knowledge/linke-turbidity-factor) anymore?

wholmgren, 2021-08-28:
Thanks @kanderso-nrel for the careful comparisons!

I think we should switch to `h5py` before releasing 0.9. I don't see any value in adding the netcdf4 layer on top of the hdf5 file for this data set. I also don't see any reason use xarray for this simple file and read operation. 

Two things that people probably already know but I feel like are not really addressed in some of the discussion above: 
1. netcdf4 uses hdf5 -- a netcdf4 file is a hdf5 file organized in a particular way plus extra metadata  
2. `pip install xarray[io]` brings in h5py along with netcdf4 and h5netcdf4, see details below

Looks like we only went with pytables because that's what pandas used and we didn't put much more thought into it: https://github.com/pvlib/pvlib-python/issues/437#issuecomment-376657958. 

<details>

```
$ pip install xarray[io]
$ pip list
Package            Version
------------------ -------------------
affine             2.3.0
appdirs            1.4.4
asciitree          0.3.3
attrs              21.2.0
beautifulsoup4     4.9.3
certifi            2021.5.30
cffi               1.14.6
cfgrib             0.9.9.0
cftime             1.5.0
charset-normalizer 2.0.4
click              8.0.1
click-plugins      1.1.1
cligj              0.7.2
docopt             0.6.2
eccodes            1.3.3
fasteners          0.16.3
findlibs           0.0.2
fsspec             2021.7.0
h5netcdf           0.11.0
h5py               3.4.0
idna               3.2
Jinja2             3.0.1
MarkupSafe         2.0.1
netCDF4            1.5.7
numcodecs          0.9.0
numpy              1.21.2
packaging          21.0
pandas             1.3.2
pip                21.2.4
pooch              1.5.1
pycparser          2.20
Pydap              3.2.2
pyparsing          2.4.7
python-dateutil    2.8.2
pytz               2021.1
rasterio           1.2.6
requests           2.26.0
scipy              1.7.1
setuptools         52.0.0.post20210125
six                1.16.0
snuggs             1.4.7
soupsieve          2.2.1
urllib3            1.26.6
WebOb              1.8.7
wheel              0.37.0
xarray             0.19.0
zarr               2.9.3
```

</details>

PR #1299 (2021-08-30): Replace tables/pytables with h5py
 - [x] Closes #1252
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~Tests added~
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue: [... cut]
closed 2021-08-30 by wholmgren