pvlib/pvlib-python #1260: "latest" docs don't make it clear that they contain unreleased features

kanderso-nrel, 2021-07-13:
**Is your feature request related to a problem? Please describe.**
Someone on stackoverflow tried using the example code in our `latest` docs with pvlib 0.8.1 because they didn't realize the feature wasn't in a stable release yet:  https://stackoverflow.com/questions/68357949/cant-find-pvlib-pvsystem-array

**Describe the solution you'd like**
A warning banner at the top of the pages at `https://pvlib-python.readthedocs.io/en/latest/` saying something like "this page might use unreleased features; click here to view documentation for the current pvlib release".  One straightforward method described [here](https://michaelgoerz.net/notes/showing-a-warning-for-the-latest-documentation-on-readthedocs.html) looks like this:

[image]

**Describe alternatives you've considered**
Maybe just disable the "latest" docs altogether?  If it's for unreleased features then what's the value in making it public?

AdamRJensen, 2021-07-13:
Recently, I have been finding myself often referring to new functions in the latest release, e.g., sending links to people. So I definitely think it has some usage. It's nice being able to somewhere seeing the current GitHub version, which I guess what 'latest' is.

As long as it defaults to 'stable', then I don't see an issue also having 'latest'. Clearly, whomever posted the question must have been peeking around the documentation in order to find the 'latest'.

I'm +1 for the banner solution!

wholmgren, 2021-07-14:
+1 for the banner, but after 0.9.

PR #1346 (2021-12-10): Add version warning banner for "latest" sphinx docs
 - [x] Closes #1260
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~Tests added~
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - ~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue  [... cut]
closed 2021-12-10 by kanderso-nrel