pvlib/pvlib-python #1263: module_height documentation

wholmgren, 2021-07-27:
Here, I don't understand `module_height`; does it not change with tracker rotation? Maybe `axis_height` and an internal calculation would produce `module_height` as an attribute, not an input parameter. In `FixedMount`, `module_height` is merely awkward, as the measurement that seems more likely to be at hand is the height of the bottom of the module above ground.

I'm OK merging this as is and improving in later PRs.

_Originally posted by @cwhanse in https://github.com/pvlib/pvlib-python/pull/1176#discussion_r675072527_

Lakshyadevelops, 2023-03-19:
aren't panels connected on their mid of face. Which would then  say height of module at mid remain constant if any rotation would occur rather than the bottom of it.
[image]
I would recommend to close this issue.

Totally noob so correct me if I am wrong.

cwhanse, 2024-11-11:
Closing because 1) the current documentation is clear and unambiguous and 2) lack of comment on the definition of this term since it was added.

closed 2024-11-11 by cwhanse

engineeringelectrical, 2025-03-10:
does module height work for monofacial panels?