pvlib/pvlib-python #1265: Server Error (400: Variable: Total_cloud_cover_low_cloud_Mixed_intervals_Average is not contained in the requested dataset)

ishaannair, 2021-07-28:
**Describe the bug**
    HTTPError: Error accessing https://thredds.ucar.edu/thredds/ncss/grib/NCEP/GFS/Global_0p5deg/Best?var=Total_cloud_cover_entire_atmosphere_Mixed_intervals_Average&var=u-component_of_wind_isobaric&var=v-component_of_wind_isobaric&var=Total_cloud_cover_boundary_layer_cloud_Mixed_intervals_Average&var=Total_cloud_cover_low_cloud_Mixed_intervals_Average&var=Downward_Short-Wave_Radiation_Flux_surface_Mixed_intervals_Average&var=Temperature_surface&var=Wind_speed_gust_surface&var=Total_cloud_cover_convective_cloud&var=Total_cloud_cover_middle_cloud_Mixed_intervals_Average&var=Total_cloud_cover_high_cloud_Mixed_intervals_Average&time_start=2021-07-28T00%3A00%3A00-07%3A00&time_end=2021-08-04T00%3A00%3A00-07%3A00&longitude=-110.9&latitude=32.2&vertCoord=100000&accept=netcdf
    Server Error (400: Variable: Total_cloud_cover_low_cloud_Mixed_intervals_Average is not contained in the requested dataset)

**To Reproduce**
Steps to reproduce the behavior: follow the steps in the website :
https://pvlib-python.readthedocs.io/en/stable/forecasts.html
1. from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP
2. model = GFS() dataFrame = model.get_processed_data(latitude, longitude, start, end)

**Expected behaviour**
a return data frame of weather forecasts between the start and end DateTime

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Versions:**
pvlib.__version: 0.9.0a7
python: 3.8.3

kanderso-nrel, 2021-07-28:
This seems like the same error as #1202 and should already be fixed in the pre-releases on pypi -- are you sure you're actually using pvlib `0.9.0a7`?  Maybe you installed that pre-release in a different environment from the one you're using for this notebook?  I do not get the error with `0.9.0a7`:

```python
In [1]: import pandas as pd
   ...: import matplotlib.pyplot as plt
   ...: import datetime
   ...: from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP
   ...: latitude, longitude, tz = 32.2, -110.9, 'US/Arizona'
   ...: start = pd.Timestamp(datetime.date.today(), tz=tz)
   ...: end = start + pd.Timedelta(days=7)
   ...: irrad_vars = ['ghi', 'dni', 'dhi']
   ...: model = GFS()
   ...: raw_data = model.get_data(latitude, longitude, start, end)
   ...: print(raw_data.head())
C:\Users\KANDERSO\Software\Anaconda3\envs\pvlib-dev\lib\site-packages\pvlib\forecast.py:21: UserWarning: The forecast module algorithms and features are highly experimental. The API may change, the functionality may be consolidated into an io module, or the module may be separated into its own package.
  'The forecast module algorithms and features are highly experimental. '
                           Medium_cloud_cover_middle_cloud_Mixed_intervals_Average  ...  u-component_of_wind_isobaric
2021-07-28 09:00:00-07:00                                          76.800003        ...                     -0.711567
2021-07-28 12:00:00-07:00                                          58.099998        ...                     -0.979160
2021-07-28 15:00:00-07:00                                          10.100000        ...                     -1.634575
2021-07-28 18:00:00-07:00                                           5.900000        ...                     -1.343711
2021-07-28 21:00:00-07:00                                           2.300000        ...                      1.505339

[5 rows x 11 columns]

In [2]: import pvlib; pvlib.__version__
Out[2]: '0.9.0-alpha.7'
```

ishaannair, 2021-07-30:
Hi,
It would seem that my notebook was working in the older version of the pvlib (0.8.0) compared to the one I had installed. After spotting the problem I have successfully got it to work with the latest version, and I am able to get all the data. 

Thank you for the help

closed 2021-07-30 by wholmgren