pvlib/pvlib-python #1269: Transformers

mikofski, 2021-07-31:
**problem**
As an energy analyst, I need to know energy at point of interconnect,  and that includes detailed transformer efficiency and losses,  as well as ac collection losses and curtailment

**solution**
A transformer load / no-load model

**alternatives**
Roll your own,  but then this risks everyone using slightly different, possibly flawed models, precisely one of the issues pvlib tries to solve

**Additional context**
* Related to #1020 and maybe #96
* a reverse model would be useful for battery storage
* copy or use pysam?

kurt-rhee, 2024-05-10:
Hey @mikofski, 

@kevinsa5 and @AdamRJensen brought this up during the PVPMC pvlib session.  I'd like to take a stab at it if there are no objections.  PlantPredict uses this very simple model from McCann, Lawrence , ABB Electrical Transmission and Distribution Reference, P.101.

https://terabase.atlassian.net/servicedesk/customer/portal/3/article/1292304413

kurt-rhee, 2024-05-10:
I would also like to write a blog post that documents how to create a pull request for a pvlib upgrade using the transformer model as a real world example.

mikofski, 2024-05-10:
If it helps I did some nuts & bolts talks at UC BIDS a few years back:
- https://bids.github.io/dats/posts/2017-10-04-github-oss-f17.html
- https://bids.github.io/dats/posts/2018-09-10-github-oss.html

kurt-rhee, 2024-05-11:
Mark those are awesome references, I will definitely riff off of them.

kandersolar, 2024-05-13:
@kurt-rhee that would be great!  FYI I skimmed around a bit in the [reference](https://archive.org/details/abb-electrical-transmission-and-distribution-reference-book-1997/page/101/mode/2up?view=theater) but did not find the equation from the PlantPredict docs.  For the pvlib function, we should cite a particular equation number or something to make it easier to find.

mikofski, 2024-05-14:
[here](https://myworkspace.dnv.com/download/public/renewables/solarfarmer/manuals/latest/CalcRef/Theory/TransformerEfficiencyModel.html) is the SolarFarmer equation. I don’t know the reference. Is it similar to Plant Predict?

kurt-rhee, 2024-05-14:
@kandersolar That is unfortunate, I looked through it as well and there is no mention of the equations that are in PlantPredict's documentation.  I've also asked around internally and none of our EE's are familiar with the equation.  Looking at the commit history, the function was last edited back in 2017 so whichever performance engineer wrote the spec for that code is likely long gone.  

As for the path forward, I could upload the PlantPredict documentation to zenodo, or if that isn't sufficient for this type of work, I have reached out to First Solar to see if any of their engineers are familiar with the function.

kurt-rhee, 2024-05-14:
@mikofski Looks like SolarFarmer and PlantPredict have different models for transformer losses.

[image]

mikofski, 2024-05-15:
@kurt-rhee I did a test and actually they’re identical! 
[Transformers.xlsx](https://github.com/pvlib/pvlib-python/files/15316494/Transformers.xlsx)
 
They’re just algebraic manipulations of each other. Now who knows the reference? Is this just common power electronics? Maybe any textbook?

mikofski, 2024-05-15:
@kandersolar that’s perfect! @kurt-rhee may I suggest moving this proof into a documentation theory section, similar to single diode but called “transformer” and then no need to reference either plant predict or SolarFarmer? The formula above is identical to the one in SF docs which I confirmed is the same as plant predict

mikofski, 2024-05-16:
Robots in disguise

adriesse, 2024-05-16:
I think it would be best to reference an introductory electrical engineering text for the simplification of the losses as:

- constant core losses, plus
- quadratic winding losses

I've thrown out my textbooks decades ago, but I assume they still teach it so I hope some young engineer can suggest one.

The total loss is a vertically shifted parabola. Spec sheets give two points on that parabola, so you use these points to solve for the two coefficients of the parabola/transformer model, which we could call ```k_core``` and ```k_winding```.

I think that all the calculations that go directly from specs to simulation needlessly and unfortunately obfuscate what is actually happening.

kurt-rhee, 2024-05-16:
@mikofski  and @kandersolar 

Seriously bravo thank you for all of the work you guys did.

echedey-ls, 2024-05-16:
> I hope some young engineer can suggest one.

It's my time to shine! I'll drop my two cents here: the only English book I see recommended in the subject I had regarding them, is `Electric machinery fundamentals. Stephen J. Chapman.`. Luckily it's easy to find freely on the Internet. Relevant section is 2.7, subsection "Transformer Efficiency", first equation is 2-65, p. 122 of the PDF, 102 of the textbook. Maybe it's a bit old, but I think it does the trick.

kurt-rhee, 2024-05-17:
@echedey-ls Awesome!

http://powerunit-ju.com/wp-content/uploads/2016/11/Book-Fundamentals-of-Electric-Machinery-by-Chapman-4ed.pdf

Here is a link to the textbook I found online.

[image]

mikofski, 2024-05-17:
@echedey-ls & @kurt-rhee sorry I don’t understand the textbook reference. I can’t see how to manipulate it to the same form as @kandersolar derivation, Plant Predict, or SolarFarmer.

echedey-ls, 2024-05-17:
I see your point @mikofski . I understood from @adriesse comment were looking for a didactic reference, not an explicit reference with the same equations. Textually, the book states some losses are constant and other losses are quadratic with the current. The following changes explain @kandersolar first equation:

$$ P_{in} = P_{Cu} + P_{core} +  V_S I_S cos \theta = P_{Cu} + P_{core} + P_{out} $$

$P_{Cu}$ relationship with output power is:

$$ P_{Cu} = I_{out}^2 R = \left( \frac{P_{out}}{V_{out}} \right) ^2 R \stackrel{V_{out}\approx constant}{=} P_{out}^2 \cdot k_{Cu}$$

Notation is different; how to determine $L_{full\,load} \enspace\text{or}\enspace k_{Cu}$ is a matter of solving for it in the first equation.

Thanks for putting the image @kurt-rhee , sometimes I forget the most obvious features of github 💀

kandersolar, 2024-05-22:
> I think that all the calculations that go directly from specs to simulation needlessly and unfortunately obfuscate what is actually happening.

I do not see a way to simplify or meaningfully separate the math shown above.  If the copper loss was defined to be proportional to $P_{in}^2$ then the equations would surely be much cleaner.  But the only way to arrive at the SolarFarmer and PlantPredict model was to assume that the copper loss was instead proportional to $P_{out}^2$, requiring the nasty quadratic formula to solve for $P_{out}$.

If you have a cleaner way, I would be quite happy to see it!

adriesse, 2024-05-23:
I'm afraid I've reached my pro bono limit on this one.  Perhaps just give it another name like "plantpredict_solarfarmer" rather than "simple"!  :)

kurt-rhee, 2024-05-23:
@adriesse @kandersolar  how about something like "abb_efficiency" so that it related to the reference?  I don't think that PlantPredict or SolarFarmer should get credit for the model since neither of us came up with it.  I'm sure @mikofski would agree.

cwhanse, 2024-05-23:
+1 for `abb_efficiency` if we change the reference to ABB's 1997 publication, which is a derivative work of the 1950s Westinghouse book.

mikofski, 2024-05-24:
Works for me

closed 2024-06-19 by kandersolar

mikofski, 2024-06-19:
Thank you so much @kurt-rhee , @echedey-ls , @kandersolar , @adriesse , & everyone who helped on this. It feels so gods to see this issue/feature-request closed after 3 years! Amazing!

PR #2053 (2024-06-19): Add a simple transformer effficiency model to pvlib

 - [x] Closes #1269
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHu [... cut]