pvlib/pvlib-python #1270: GitHub Discussions

AdamRJensen, 2021-08-04:
I recently stumbled upon the [xarray package which uses GitHub Discussions](https://github.com/pydata/xarray/discussions) very well.

It seems that within pvlib we often tend to make discussions in issues, where eventually someone ends up referring to a comment five comments back and things get hard to follow. With GitHub discussions comments are not necessarily linear in time, as you have the option of commenting on comments (sorta like a sub-thread).

Some of the benefits of GitHub Discussions:
* Separates issues (e.g., feature requests, bug fixes) from discussions
* Allows for cross-referencing 
* Supports tagging of discussion for each organization and searching

I think having GitHub Discussion would urge more discussion, e.g., of the roadmap, and even having Q/A where you can directly reference the functions being used.

AdamRJensen, 2021-08-04:
I would tag @pvlib/pvlib-maintainer, however, it doesn't seem to let me do that..

wholmgren, 2021-08-07:
It appears that tag is restricted to members of the pvlib organization and that cannot be changed. So yes, we should stop encouraging people to use it. 

I've seen plenty of github discussions go off the rails but I am fine with experimenting with them.

mikofski, 2021-08-08:
I'm fine with discussions

wholmgren, 2021-08-09:
Closed by https://github.com/pvlib/pvlib-python/discussions/1276

closed 2021-08-09 by wholmgren