pvlib/pvlib-python #1278: Add a model for spectral corrections

Jacc0027, 2021-08-11:
**Additional context**
First of all, I introduce myself, my name is Jose Antonio Caballero, and I have recently finished my PhD in photovoltaic engineering at the University of Jaén, Spain.

I have developed a python script to apply spectral corrections as a function of AM, AOD, PW based on this work (https://doi.org/10.1109/jphotov.2017.2787019).

We have found that in pvlib there is already a similar methodology developed by First solar, in which the spectral corrections are based only on the AM and PW parameters, so we intend to include our proposed method in pvlib in a similar way.

As an example, I attach the code developed in python (.zip file) to estimate the spectral effects related to different flat photovoltaic technologies from the AM, AOD and PW parameters included in a .csv file.
[PV-MM-AM_AOD_PW_data.csv](https://github.com/pvlib/pvlib-python/files/6970716/PV-MM-AM_AOD_PW_data.csv)
[PV_Spectral_Corrections.zip](https://github.com/pvlib/pvlib-python/files/6970727/PV_Spectral_Corrections.zip)

Kind regards

cwhanse, 2021-08-11:
Hello @Jacc0027 and welcome to this community.

I would welcome submission of a function implementing that model. It differs from the existing functions by including parameters for perovskite cells.

It would need to be coded as a function similar to [first_solar_spectral_correction](https://github.com/pvlib/pvlib-python/blob/f318c1c1527f69d9bf9aed6167ca1f6ce9e9d764/pvlib/atmosphere.py#L324) using the same parameter names, where possible.

The script isn't in a function format. The time series in the .csv file appear to be related to the SMARTS simulations used to set up this model, is that correct? If so, then these data don't need to come into pvlib.

Jacc0027, 2021-08-11:
My pleasure @cwhanse , thank you very much for the welcome.

No problem, I can adapt the code in a similar way to the one you attached, and assuming the same variables.

Yes, you are right, the script I attached is not made as a function, it was just a sample to verify the results. And again you are right, the time series in the .csv file are obtained through SMARTS simulations

Therefore, I will attach in this thread the modified code as we have just agreed.

Jacc0027, 2021-08-13:
Hi @cwhanse , please find attached the updated script, in which we have tried to meet the requested requirements.

I look forward to any comments you may have.

[AM_AOD_PW_spectral_correction.zip](https://github.com/pvlib/pvlib-python/files/6984688/AM_AOD_PW_spectral_correction.zip)

adriesse, 2021-08-28:
I suggest putting some information into the doc string about the range of applicability.  Is it for clear-sky conditions only, or all sky?  Is it for 30-degree tilt only, a range of tilts, or any orientation?

Jacc0027, 2021-09-04:
Hello Anton, 

the proposed experimental equations are applicable in all sky conditions.

The fit coefficients were obtained experimentally with integrated and spectral irradiance measurements with an inclination of 30º above the horizontal. Strictly speaking, this would be the valid tilt for the model.

Regarding the orientation, the experimental installation was completely oriented towards the south, so it can be applied considering this scenario and subsequently applying the corresponding adjustment for orientation losses associated with a different orientation to the one considered in the model. This methodology is usually applied in the modelling of PV systems independently of this spectral adjustment.

The limitations of the model remain to be verified, as indicated in both,  the proposed python script and the published paper. It remains to be seen how the adjustment coefficients are modified in locations with extreme climates where AOD and PW values are frequently high.
In any case, we are firmly convinced that this model performs better than others that do not consider AOD as an input parameter, since it is an important factor that affects short wavelengths significantly.

I remain at your disposal in case of any further questions.

PR #1296 (2023-06-23): Implement Caballero et al. spectral factor model
 - [x] Closes #1278 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub I [... cut]
closed 2023-06-23 by kandersolar