pvlib/pvlib-python #1302: Installing 0.9.0 wheel on python 3.6 does not install dataclasses, causes import failure

kanderso-nrel, 2021-09-02:
**Describe the bug**
Installing pvlib 0.9.0 using the wheel currently on PyPI does not install dataclasses, causing pvlib to fail to import on python 3.6.  I think this is because the wheel builder workflow uses python 3.8, so it ignores the dataclasses dependency when building the wheel (which is supposed to be universal across all our supported python versions):

https://github.com/pvlib/pvlib-python/blob/518cc35567baa15ae203bb8537452927ae10c7a7/.github/workflows/publish.yml#L24

https://github.com/pvlib/pvlib-python/blob/518cc35567baa15ae203bb8537452927ae10c7a7/setup.py#L49-L50

Opening the wheel (it's just a zipfile) and viewing the `pvlib-0.9.0.dist-info/METADATA` file, there is no mention of dataclasses.  Note that installing 0.9.0 from source (`pip install pvlib --no-binary pvlib`) works as intended because it is running setup.py locally in python 3.6.  Our CI never has this kind of python version mismatch between what was used to build the wheel and what was used to run the tests, so it didn't catch this. 

**To Reproduce**
```
$ conda create -n py36test python=3.6
$ conda activate py36test
$ pip install pvlib==0.9.0
$ python -c "import pvlib"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\KANDERSO\Software\Anaconda3\envs\py36test\lib\site-packages\pvlib\__init__.py", line 3, in <module>
    from pvlib import (  # noqa: F401
  File "C:\Users\KANDERSO\Software\Anaconda3\envs\py36test\lib\site-packages\pvlib\ivtools\__init__.py", line 7, in <module>
    from pvlib.ivtools import sde, sdm, utils  # noqa: F401
  File "C:\Users\KANDERSO\Software\Anaconda3\envs\py36test\lib\site-packages\pvlib\ivtools\sdm.py", line 16, in <module>
    from pvlib.pvsystem import calcparams_pvsyst, singlediode, v_from_i
  File "C:\Users\KANDERSO\Software\Anaconda3\envs\py36test\lib\site-packages\pvlib\pvsystem.py", line 14, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
```

**Additional context**
Noticed in https://github.com/NREL/rdtools/pull/290

Possible solutions:
- Switch the deploy workflow to use python 3.6, which will end up requiring `dataclasses` for all python versions
- Upgrade `dataclasses` to be required for all versions in `setup.py`
- Build separate wheels for each python version, maybe with [cibuildwheel](https://github.com/pypa/cibuildwheel)?

wholmgren, 2021-09-02:
I manually built the wheel using python 3.6 and uploaded to testpypi. It seems to work.

```
$ python setup.py sdist bdist_wheel --python-tag py36
$ twine upload --repository testpypi dist/pvlib-0.9.0-py36-none-any.whl
# new session
$ conda create -n pvlib36test python=3.6
$ conda activate pvlib36test
$ pip install --index-url https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple/ pvlib
$ python
>>> import pvlib
>>> pvlib.__version__
'0.9.0'
```

As expected, running `pip install --index-url https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple/ pvlib` in a python 3.8 environment gives me the pvlib 0.9 from primary pypi.

Should I go ahead and upload the py36 whl to primary pypi too or do you want to try cibuildwheel?

kanderso-nrel, 2021-09-02:
Fine with me to upload the 36 wheel just to get this fixed in the short term.  Will it conflict with the `pvlib-0.9.0-py3-none-any.whl` already there?  I don't know how the wheel preference logic works when more than one wheel would work.

Long run I'll volunteer to set up cibuildwheel if we think that's the way to go.

wholmgren, 2021-09-02:
Uploaded. See [here](https://www.python.org/dev/peps/pep-0425/#id1) for documentation on order of preference.

wholmgren, 2021-09-02:
I believe a simple solution is to remove 

https://github.com/pvlib/pvlib-python/blob/518cc35567baa15ae203bb8537452927ae10c7a7/setup.py#L48-L50

and add the following to `INSTALL_REQUIRES`: `'dataclasses; python_version < "3.7"'`.

I did that, rebuilt the wheel, and inspected the MANIFEST. It contained `Requires-Dist: dataclasses ; python_version < "3.7"`. `pip install dist/pvlib-0.9.0+0.g518cc355.dirty-py3-none-any.whl` correctly installed `dataclasses` in a python 3.6 environment. The same command in a python 3.8 environment did not attempt to install `dataclasses`.

In any case, the conda-forge package is also broken on 3.6.

kanderso-nrel, 2021-09-02:
Nice.  Would it make sense to "re-release" 0.9.0 to PyPI using that syntax, or are you happy with the state of PyPI as-is?  https://github.com/NREL/rdtools/pull/290 is happy with it, so maybe no need to change anything.

I assume the `pvlib` conda channel has the same issue, so I'll take a look at that and conda-forge now.

wholmgren, 2021-09-02:
I'm happy with the current state so I'd say leave it alone.

kanderso-nrel, 2021-09-02:
Ok, the `pvlib` and `conda-forge` channels have new builds (build `1`) that work for me on both 3.6 and 3.7, so I think everything is fixed.  If anyone discovers that's not the case, please let me know.

PR #1422 (2022-03-17): Fix conditional dependency on dataclasses
 - [x] Closes #1302
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub I [... cut]
closed 2022-03-17 by wholmgren