pvlib/pvlib-python #1309: Add fitting function for Anton's inverter model

Peque, 2021-09-15:
pvlib includes de `inverter.adr()` model, but it still does not provide a way to infer the required parameter for the model given, for example, some known datasheet parameters. The fitting function would be a great addition to be able to work with Anton's model.

I think @adriesse had a MATLAB implementation, so this issue is to discuss its addition and track progress when/if Anton finds some time to implement the fitting function in Python. :blush:

adriesse, 2021-10-17:
I have finally taken some time to work in this topic, which has been at the bottom of my to-do list for too long.  You can now find a ready-to-use function to convert from Sandia model parameters to ADR model parameters here:

https://github.com/adriesse/pvpltools-python

There are also helper functions that can be used with suitable measured data, the CEC matrix, or other tabulated specifications.

Related issue: https://github.com/pvlib/pvlib-python/issues/317.

adriesse, 2021-11-17:
Does this seem useful to anyone?

cwhanse, 2021-11-17:
> Does this seem useful to anyone?

For this specific issue, the two `fit_xxx` functions in `power_conversion.py` would be valuable to add to `pvlib.inverter` if you are offering that code.

adriesse, 2021-11-22:
It's written with pvlib in mind, but I'm still thinking about some improvements.