pvlib/pvlib-python #1313: Integrate Solcast API into iotools

mikofski, 2021-09-24:
**Is your feature request related to a problem? Please describe.**
Get irradiance via Solcast API
https://solcast.com/solar-data-api/api/

**Describe the solution you'd like**
Use iotools to get irradiance data from Solcast

**Describe alternatives you've considered**
Solcast already has an API toolkit. Is irons redundant?

**Additional context**
* https://docs.solcast.com.au/#introduction
* https://solcast.com/solar-data-api/api-toolkit/
* icymi @AdamRJensen

williamhobbs, 2022-01-07:
This could be expanded to include other APIs, such as Solargis (https://solargis.com/products/api/data-delivery-web-service/overview) and SolarAnywhere (https://developers.cleanpower.com/irradiance-and-weather-data/).

kanderso-nrel, 2022-01-07:
A bit of relevant discussion here: https://github.com/pvlib/pvlib-python/discussions/1310

mikofski, 2022-01-07:
Hi @kanderso-nrel is this issue superceded by the Engerer PR #1366 ?

kanderso-nrel, 2022-01-07:
@mikofski Solcast and Engerer don't seem related to me, what's the connection?  Or maybe you meant another issue?

mikofski, 2022-01-08:
[Nick Engerer is the CTO and a co-founders of Solcast](https://www.linkedin.com/in/dr-nick-engerer/), so I thought maybe their software uses his methods? But TBH that's just an assumption.

kanderso-nrel, 2022-01-10:
Ha, of course!  I think #1366 does not supercede this one: decomposition function in `pvlib.irradiance` vs data fetching function in `pvlib.iotools`.

PR #1875 (2023-12-20): add functionality to load Solcast API data to `iotools`
 - [x] Closes #1313 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the G [... cut]
closed 2023-12-20 by AdamRJensen