pvlib/pvlib-python #1334: Increase the usage of augmented assignment statements

elfring, 2021-10-31:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

Would you like to integrate anything from a transformation result which can be generated by a command like the following?
(:point_right: Please check also for questionable change suggestions because of an evolving search pattern.)
```
[Markus_Elfring@fedora lokal]$ perl -p -i.orig -0777 -e 's/^(?<indentation>\s+)(?<target>\S+)\s*=\s*\k<target>[ \t]*(?<operator>[+\-%&|^@]|\*\*?|\/\/?|<<|>>)/$+{indentation}$+{target} $+{operator}=/gm' $(find ~/Projekte/pvlib-python/lokal -name '*.py')
```

kanderso-nrel, 2021-11-01:
Hi @elfring, I think we're generally open to improving our codebase, but I'm not sure perl regexes would be the tool of choice to do it.  Seems like a proper linter, or something like [black](https://github.com/psf/black), would make more sense.  What do you think?

Also, any reason why you're interested in augmented assignment statements in particular?

Probably worth pointing out that there is a slim chance that blindly replacing `x = x+y` with `x += y` could break something since `+=` generally edits in-place while `+` creates a new object (though of course it depends on how each object defines `__add__` and `__iadd__`).

elfring, 2021-11-01:
>…, but I'm not sure perl regexes would be the tool of choice to do it.

The programming language “Perl” provides special support for the application of regular expressions.
Thus it can look convenient to use it for some data processing approaches.

> Seems like a proper linter, …, would make more sense.

Such software development tools are evolving in some directions.

>Also, any reason why you're interested in augmented assignment statements in particular?

:thought_balloon: I hope that some information ([which was published also together with release notes for Python 2](https://docs.python.org/3/whatsnew/2.0.html#augmented-assignment "Augmented assignments")) can be taken better into account.
How do you think about to support another bit of collateral evolution? :thinking:

>… could break something since `+=` generally edits in-place while `+` creates a new object

:thought_balloon: This technical detail influences run time characteristics in potentially desirable ways, doesn't it?

>(though of course it depends on how each object defines `__add__` and `__iadd__`)

Would you like to [clarify your development concerns in more detail](https://docs.python.org/3/reference/datamodel.html#object.__iadd__ "Documentation for augmented arithmetic assignments")? :thinking: