pvlib/pvlib-python #1340: Transposition model with multiple tilts

kurt-rhee, 2021-11-12:
**Is your feature request related to a problem? Please describe.**
I would like to calculate transposition given an azimuth, a tilt angle, and a rotation angle.  This may occur if you have a single axis tracker which has a non-zero, non-trivial change in elevation along the tracker axis as well as a phi rotation about the torque tube.  My current understanding is that pvlib assumes that a tracker lies in a flat plane.

**Describe the solution you'd like**
I would like a function within pvlib.irradiance to be able to calculate this.  

**Describe alternatives you've considered**
I have not considered alternatives, I am willing to help implement this with some guidance.

**Additional context**
No additional context.

kurt-rhee, 2021-11-12:
For additional context:

In a right handed coordinate system where the index finger (x) points to the east, the middle finger points away from the observer at origin (y), and the thumb points upwards towards zenith (z).  I would like a transposition model that can accept:
- solar module rotations about the z-axis (thumb) to model azimuth
- solar module rotations about the y-axis (middle finger) to model the tilt of the module about a single axis tracker
- solar module rotations about the x-axis (index finger) to model the tilt of the tracker itself

kanderso-nrel, 2021-11-12:
Howdy @kurt-rhee, pvlib models SAT as horizontal by default, but tilted-axis SAT is supported via `pvlib.tracking.singleaxis`'s `axis_tilt` parameter.  The `surface_tilt` and `surface_azimuth` values returned will have the tilted axis baked in, so they should be appropriate to pass along to the functions in `pvlib.irradiance` as-is.  I don't think a fancy transposition model is really needed since transposition only needs tilt & azimuth and doesn't care where the values came from (tracker or otherwise).

Hopefully I've not misunderstood your point, please clarify if I'm misunderstanding :)

kanderso-nrel, 2021-11-12:
Illustrating the "baked in" point, here's a short example showing how `axis_tilt` affects the simulated POA irradiance.  If the trackers were horizontal then POA would touch GHI at solar noon, but in this case the tracker axis is tilted facing to the south a bit, so `surface_tilt` is nonzero at solar noon and POA and GHI don't touch.

```python
import pvlib
import pandas as pd
import matplotlib.pyplot as plt

times = pd.date_range('2019-01-01', '2019-01-02', freq='1min', tz='Etc/GMT+5')
location = pvlib.location.Location(40, -80)
solpos = location.get_solarposition(times)
clearsky = location.get_clearsky(times)
tracker = pvlib.tracking.singleaxis(solpos.apparent_zenith, solpos.azimuth, axis_tilt=5, axis_azimuth=180)
poa = pvlib.irradiance.get_total_irradiance(tracker.surface_tilt, tracker.surface_azimuth,
                                            solpos.apparent_zenith, solpos.azimuth,
                                            clearsky.dni, clearsky.ghi, clearsky.dhi)
clearsky.ghi.plot(label='GHI')
poa['poa_global'].plot(label='POA')
plt.ylabel('Irradiance [W/m2]')
plt.legend()
```
[image]

kurt-rhee, 2021-11-12:
hey @kanderso-nrel, nice to talk to you and really enjoyed your papers on slope aware backtracking they were really helpful and easy to understand.

As for the original question, your answer makes total sense and I see now that th eoutput of pvlib.tracking.singleaxis gives separate values for tracker_theta and surface tilt when the tracker axis tilt is != 0.

Would you happen to have any visuals or docs for me to understand the relationship between tracker theta, tracker axis tilt and surface tilt for my implementation?

kanderso-nrel, 2021-11-13:
Thanks!

Section 2 of [Marion & Dobos](https://www.nrel.gov/docs/fy13osti/58891.pdf) is certainly relevant, although I personally find that diagram a bit difficult to grasp.  I'm afraid I can't think of any others off the top of my head.

kurt-rhee, 2021-11-13:
Thank you!  I'll give it a read over the weekend.

closed 2021-11-13 by kurt-rhee