pvlib/pvlib-python #1342: gti_dirint raises 'UnboundLocalError: local variable 'best_ghi' referenced before assignment'

kdebrab, 2021-11-24:
**Describe the bug**
`irradiance.gti_dirint` fails for surface faced North

**To Reproduce**
```python
import pvlib
import pandas as pd
times = pd.date_range('2021-11-24', '2021-11-25', freq='h')
latitude, longitude, surface_tilt, surface_azimuth = 51.3, 4.56, 25, 360
solpos = pvlib.solarposition.get_solarposition(times, latitude, longitude)
solar_zenith, solar_azimuth = solpos['apparent_zenith'], solpos['azimuth']
ghi = pvlib.clearsky.simplified_solis(90 - solar_zenith)['ghi']
aoi = pvlib.irradiance.aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth)
pvlib.irradiance.gti_dirint(ghi, aoi, solar_zenith, solar_azimuth, times, surface_tilt, surface_azimuth)
```
raises
```
UnboundLocalError: local variable 'best_ghi' referenced before assignment
```
The error is not raised e.g. when `surface_azimuth = 180`.

**Versions:**
 - ``pvlib.__version__``: 0.9.0
 - ``pandas.__version__``: 1.3.4
 - python: 3.9.4.final.0

kdebrab, 2021-11-24:
The example is not fully correct (I should have converted ghi to the plane before feeding it to gti_dirint), but that doesn't really matter for the bug report.

In this case, the plane doesn't see the sun during the times, so diriving the components would be trivial (it's all diffuse), but, in any case, the algorithm should not raise an error.

cwhanse, 2021-11-24:
Confirmed. The problem appears to be that `_gti_dirint_lt_90` is always called, without checking if any `aoi` value is less than 90. The test for `gti_dirint` doesn't hit this corner case.

Maybe put an `if` block around this [line](https://github.com/pvlib/pvlib-python/blob/aba071f707f9025882e57f3e55cc9e3e90e869b2/pvlib/irradiance.py#L1906).

wholmgren, 2021-11-24:
It's not as simple as putting an `if aoi_lt_90.any()` around `_gti_dirint_lt_90`. `_gti_dirint_gte_90` requires the `best_kt_prime` determined in `_gti_dirint_lt_90`. I don't think we can make this work without going beyond what's in the Marion paper. I suggest we instead return an informative error if insufficient data is supplied with aoi < 90.

adriesse, 2025-01-01:
How about just assigning some default values to 'best_ghi' earlier in the code?

cwhanse, 2025-01-02:
I'm open to suggestions. `best_ghi` is the GHI returned from the optimization. Where that can't be done, maybe return `np.nan`? That seems less helpful to me than interrupting the calculation and informing the user.

adriesse, 2025-01-05:
`np.nan` seems ok to me.  Either way, the Notes section could explain the problem a bit.

PR #2347 (2025-02-12): Raise helpful exceptions for irradiance.gti_dirint
 - [x] Closes #1342
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [x] Tests added
 - ~~[] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to th [... cut]
closed 2025-02-12 by kandersolar