pvlib/pvlib-python #1345: CEC module library is behind current version in SAM

kanderso-nrel, 2021-11-29:
As discussed in https://github.com/pvlib/pvlib-python/discussions/1344, the CEC module table shipped with pvlib (`SAM 2018.11.11 r2`, `1/3/2019`) is behind the latest version shipped with SAM (`SAM 2020.2.29 r3`, `11/25/2020`).

@mikofski in https://github.com/pvlib/pvlib-python/issues/440 you brought up pvfree -- any thoughts there this time around?

mikofski, 2021-11-29:
It's possible to update pvfree by clicking the admin pulldown, selecting upload, and uploading the desired csv. IMO some advantages of using pvfree or something similar:
1. library updates don't require a pvlib re-release if they two are decoupled
2. pvfree db doesn't allow integrity or validation errors, meaning, you can't upload the same file twice or put a string in a number field
3. human readable, searchable, filterable
4. API has built in filters that make it easier to locate desired module(s) or inverter(s)
5. uses off the shelf, vanilla packages (django, tastypie) and 3rd party zero-admin webhosting and deployment (github, heroku) and database tools (elephantSQL). This means pvfree has built in convenience tools like admin page, filtering, security updates, etc. which lowers maintainer burden.

mikofski, 2021-12-01:
see https://github.com/BreakingBytes/pvfree/issues/34 might need to update this table in the source, easy pr

wholmgren, 2021-12-01:
>library updates don't require a pvlib re-release if the two are decoupled

+1. But that feels like a breaking change even if it's not a real API issue, so I'll make the same suggestion for a short term solution as in #440: make a minor pvlib release with the new files AND update pvfree. 

https://github.com/pvlib/pvmodules is also relevant. I don't recall if there was a block other than lack of time. @cwhanse?

cwhanse, 2021-12-01:
>  I don't recall if there was a block other than lack of time.

That's mostly it. NREL also wants to open source the production of the CEC module file. @janinefreeman

kanderso-nrel, 2022-01-24:
> library updates don't require a pvlib re-release if the two are decoupled

> +1. But that feels like a breaking change

Perhaps that concern could be addressed by pvfree exposing multiple versions of the CEC table via a new API parameter, similar to how the PSM3 API has `tmy-2018` and such.  Then code that specifies `cecdb='2018.11.11'` (or whatever) would behave the same even after a new version of the table is added.

cwhanse, 2022-03-24:
As an interim, we could update the file distributed with pvlib-python to the current SAM release.

cwhanse, 2022-12-09:
SAM 2022.11.21 is released with updated module and inverter libraries: https://github.com/NREL/SAM/tree/patch/deploy/libraries