pvlib/pvlib-python #1347: How to get more then 30days of GFS Data?

thiagomoeng, 2021-11-30:
Hi, I'm a new pvlib user and I'm having a problem trying to request more than 30 days of GFS data. Is there any way to get/download data for the entire year?

Sample Code:

```
import pandas as pd
import matplotlib.pyplot as plt
import datetime
from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP

latitude, longitude, tz = -40, -40, 'Brazil/East'
start = pd.Timestamp(datetime.datetime(2021, 1,25, 00, 00), tz=tz)
end = pd.Timestamp(datetime.datetime(2021, 11, 25, 00, 00), tz=tz)     #start + pd.Timedelta(days=30)
irrad_vars = ['ghi', 'dni', 'dhi']
model = GFS()

raw_data = model.get_data(latitude, longitude, start, end)

print(raw_data)
```

And the result only contains data from the last 30 days as follows
`2021-10-30 21:00:00-03:00                                           0.000000                                           0.000000               ...                      0.000000                      0.000000
2021-10-31 00:00:00-03:00                                           0.000000                                           0.000000               ...                      7.795835                     -5.171963
2021-10-31 03:00:00-03:00                                           0.000000                                           0.000000               ...                      8.367566                     -5.228735
2021-10-31 06:00:00-03:00                                           0.000000                                           0.000000               ...                      8.600102                     -5.188425
2021-10-31 09:00:00-03:00                                          13.500000                                          13.500000               ...                      8.529766                     -6.115500`

williamhobbs, 2022-01-07:
I don't think that the servers that pvlib uses to pull NWP data (like GFS) serve as a long-history archive, and instead have a limited rolling window for available data, e.g., 4 weeks or 30 days. 

NOAA does use AWS for some of the NWP outputs, e.g., GEFS (GFS ensemble), https://registry.opendata.aws/noaa-gefs/, and HRRR, https://registry.opendata.aws/noaa-hrrr-pds/. I'm not sure of the status, but solarforecastarbiter was looking at pulling GEFS from AWS instead for NOMADS (default server for solarforecastarbiter, but different from what pvlib uses). See here for details: https://github.com/SolarArbiter/solarforecastarbiter-core/pull/712.

Also see:
- https://github.com/SolarArbiter/solarforecastarbiter-core/issues/559
- An example for manually pulling from NOMADS, but additional work may be required: https://github.com/wholmgren/get_nomads

AdamRJensen, 2022-07-22:
Closing this issue on the basis that the forecasting module is being deprecated.

closed 2022-07-22 by AdamRJensen