pvlib/pvlib-python #1350: account for major pvwatts changes

wholmgren, 2021-12-07:
The [SAM Version 2021.12.02 release notes](https://nrel.github.io/SAM/doc/releasenotes.html) describe major changes to the PVWatts model:

> With this release, we have updated SAM's implementation to PVWatts V8 (pvwattsv8 in SSC) with new features listed below. The online PVWatts® Calculator continues to use pvwattsv5 and will be updated later. All versions of PVWatts are available via the SAM Software Development Kit and PySAM. (729)
> 
>     Reorganize user interface to group advanced inputs into a collapsible panel.
>     Add snow loss model when weather file includes snow depth data in cm.
>     Support for bifacial modules.
>     Internal module and module thermal models based on same model as Detailed PV model.
>     Internal inverter model based on same model as Detailed PV model.
>     Inputs for monthly soiling as DC loss.

We should add a note or a warning to `pvsystem.pvwatts_dc` and `inverter.pvwatts`. And `ModelChain.with_pvwatts` is also now ambiguous.

kanderso-nrel, 2021-12-07:
Append `v5` to the function names?  So `ModelChain.with_pvwattsv5`, etc, and deprecate the originals

cwhanse, 2021-12-07:
After thumbing up, second thought: append `v5` to base layer functions but add a parameter to `ModelChain.with_pvwatts`.

adriesse, 2022-01-03:
The release note itself seems ambiguous, since the "Detailed PV Model" has several module and inverter options, and incomplete, since the parameters for those options aren't identified.  Regardless, I think it's a good idea to create and maintain ways in which pvlib users can run a version of their choice and document what that choice implies technically.

adriesse, 2022-01-04:
The following note from the SAM website may also be relevant:

> Note: The PVWatts website uses a different version numbering system than the PVWatts model. For example, the website "Version 6" uses the PVWatts Version 5 model.

adriesse, 2022-09-23:
Is the overall idea to offer a separate set of functions for each (or at least more than one) pvwatts version?

kanderso-nrel, 2022-09-24:
I think there are two independent goals here.  For one, to resolve the ambiguity of what models our current "pvwatts_dc" et al. functions implement.  For two, to make space for emulating other PVWatts versions, which yes is something I'd like to do for at least v8.

FYI I suspect much of PVWatts v8 is already implemented in pvlib (although called something else, e.g. `sam_noct` and the CEC models), but I'm waiting until they publish an official description before taking a closer look.

adriesse, 2022-09-24:
I support those goals.  Assuming that there will be several versions implemented eventually (someone might want to add older ones too) then probably there will be only a few base functions that differ from one version to the next.  Would the version suffix refer to the version where the function was introduced?  Do any of them them predate v5 currently?

spolloni, 2025-04-18:
> FYI I suspect much of PVWatts v8 is already implemented in pvlib (although called something else, e.g. `sam_noct` and the CEC models), but I'm waiting until they publish an official description before taking a closer look.

hey @kandersolar! found this issue while trying to find guidance on how to implement an approach truer to v8 than v5 with pvlib. Any tips there?

kandersolar, 2025-04-18:
Sorry, I don't have anything new to add here.  I still haven't seen a spec for PVWatts v8 (although I haven't really looked, either).

markcampanelli, 2025-08-15:
> I almost created a new issue about adding clarification to the pvwatts_dc
> documentation,
> https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.pvwatts_dc.html,
> and then I found this issue, so I'll add me thoughts here. I'd be glad to
> create a separate issue if something thinks it's worthwhile.
>
> A few suggestions about the pvwatts_dc docs:
>
>    1. The reference in the documentation is for PVWatts v5, which helps,
>    but I think the first sentence should be something like, "Implements NREL’s
>    PVWatts *(Version 5)* DC power model."
>    2. I think we should also consider adding some thing like, "This model
>    has also been referred to as the power temperature coefficient model",
>    because that's what it was called in Marion 2008 [1].
>    3. Maybe it's too nuanced, but it would be cool if we could cite the
>    origin of this temperature coefficient model. [2] has this equation but for
>    efficiency instead of power, and it references [3] as the source.
>    https://doi.org/10.1016/0038-092X(81)90220-6. Maybe we could cite [3],
>    and see if we can find when it was first applied directly to power?
>
> And as a side-note about [1], it refers to the PVWatts v1 model as "the
> PVFORM model", which supports @kandersolar
> <https://github.com/kandersolar>'s suggestion that if we ever implement
> PVWatts v1, we should call it something like pvform.
>
> [1] B. Marion, "Comparison of Predictive Models for Photovoltaic Module
> Performance," IEEE PVSC 2008,
> https://docs.nrel.gov/docs/fy08osti/42511.pdf
> [2] M.D. Siegel, S.A. Klein, W.A. Beckman, A simplified method for
> estimating the monthly-average performance of photovoltaic systems, Solar
> Energy, 1981, https://doi.org/10.1016/0038-092X(81)90220-6
> [3] Jet Propulsion Lab., A Solar Cell Array Design Handbook, Vol. II. No.
> JPL SP 43-38 (1976).
> https://ntrs.nasa.gov/api/citations/19770007250/downloads/19770007250.pdf

Will: Thanks for documenting your recent findings as recommendations here.

Does guidance exist anywhere about the known limitations of this model?

Mark Campanelli
LinkedIn <https://www.linkedin.com/in/markcampanelli/>
Intelligent Measurement Systems LLC
<https://intelligentmeasurementsystems.com>
Try PVfit <https://pvfit.app> today!

williamhobbs, 2025-08-16:
> Does guidance exist anywhere about the known limitations of this model?

@markcampanelli, the Marion 2008 reference talks a little about how the linear power temperature coefficient model doesn’t capture non-linear impact of lower efficiency at lower irradiance. 

Deville 2023 (https://doi.org/10.1002/pip.3763) is probably the most robust evaluation I’ve seen, but there could be others. Fig. 13 there does a great job illustrating the low irradiance issue. But for the climate of the test location, it doesn’t matter too much for total energy.

PR #1558 (2022-12-14): Rename `pvwatts` to `pvwattsv5` everywhere
 - [x] Closes #1350
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issu [... cut]