pvlib/pvlib-python #1358: Change add position in new pydata theme

AdamRJensen, 2021-12-16:
**Is your feature request related to a problem? Please describe.**
With the new pydata theme implemented in #1173, the ads from readthedocs have become much more prominent. E.g., the main pvlib landing page in the [latest version](https://pvlib-python.readthedocs.io/en/latest/) is dominated by the advertisement (see screenshot from a phone below):

[image]

**Describe the solution you'd like**
While there are good reasons for having the advertisement, I do not think that it should be the most dominant feature that greets users. 

One solution would be to move the advertisement to the bottom of the page. Though I have not been able to determine if this is possible with the theme. Perhaps @choldgraf could add a comment on this?

I noticed that other sites using the template also feature the ad at the top on the mobile version, e.g., [pydata theme](https://pydata-sphinx-theme.readthedocs.io/en/latest/index.html), binder, and jupyterhub.

**Additional context**
I fully support changing to the new theme btw.

wholmgren, 2021-12-16:
Of these options:

1. Pay readthedocs
2. Hack the theme
3. Build the docs in the ci, post to a github.io repository. 
4. Live with the ads

I prefer 4, 1, 3, 2. 

Would be nice if PVPMC could pay for rtd ad free hosting and, while we're at it, a domain name too. While the combined price of that is roughly $100 per year, I'm guessing the overhead in making that happen is on the order of $1000 per year, unfortunately. Maybe there's a more efficient path through numfocus or github sponsors.

choldgraf, 2021-12-16:
I think that the right long-term solution is to make the sidebar collapsible when you're on mobile, rather than having it displayed at the top of the page. Agreed it is sub-optimal

AdamRJensen, 2021-12-16:
Another option would be to move the advertisement to the right-hand side. This only would require a minor change of the sphinx settings and avoids the advertisement being placed at the top on smartphones.

In general the [left-hand side of a webpage tends to get the most attention](https://cxl.com/blog/10-useful-findings-about-how-people-view-websites/), so either way, I would favor moving the advertisement to the right.

choldgraf, 2021-12-16:
ah I just realized that you could do this already with theme configuration, by following these instructions:

https://pydata-sphinx-theme.readthedocs.io/en/latest/user_guide/sections.html#the-left-sidebar

You should try putting the `sidebar-ethical-ads` snippet in the footer:

https://pydata-sphinx-theme.readthedocs.io/en/latest/user_guide/sections.html#the-footer

does that work?

AdamRJensen, 2021-12-17:
@choldgraf I successfully moved the ad to the footer in #1359 and it worked like a charm!

I really think it's a win-win, it moves the advertisement to the bottom on smartphones and makes it less dominating in the browser version.

Would it make sense to make this the default for the theme?

choldgraf, 2021-12-17:
well we probably don't want the advertisement in the footer on a full width, since it'd generally be hidden, but I think the "hidden sidebar" that I mentioned above is a reasonable compromise

PR #1359 (2021-12-27): Move rtd advertisement to footer

 - [x] Closes #1358
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
~~- [ ] Tests added~~
~~- [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with  [... cut]
closed 2021-12-27 by kanderso-nrel