pvlib/pvlib-python #1360: ```get_pvgis_tmy``` results don't match PVGIS website

UGuntupalli, 2021-12-27:
**Describe the bug**
Data retrieved using pvlib does not match data in pvgis website 

**To Reproduce**
Steps to reproduce the behavior:
```python
import pvlib 
from pvlib.iotools import get_pvgis_tmy 

# Print version to confirm it is the latest 
print(pvlib.__version__)  # confirms it is 0.9.0 

# Define two coordinates (1 each in US and EU) 
latitude_us, longitude_us = 33.21, -95.72
latitude_eu, longitude_eu = 37.66, 13.76

# Pull TMY data for both locations 
us_pvgis_tmy, _, _, _ = get_pvgis_tmy(latitude_us, longitude_us)
eu_pvgis_tmy, _, _, _ = get_pvgis_tmy(latitude_eu, longitude_eu)

# Calculate annual GHI 
print(f''' Annual GHI at US Test site is: {us_pvgis_tmy['G(h)'].sum()/1000}''')
print(f''' Annual GHI at EU Test site is: {eu_pvgis_tmy['G(h)'].sum()/1000}''')
```

**Expected behavior**
Data download using pvlib iotools should match the original source i.e. Pvgis 

**Screenshots**

[image]

**Versions:**
 - ``pvlib.__version__``:  0.9.0
 - ``pandas.__version__``: 1.2.4 
 - python: 3.8.0 

**Additional context**
Using the Pvgis built-in TMY tool (https://re.jrc.ec.europa.eu/pvg_tools/en/#TMY), the following files were downloaded from PVGIS website for the test sites used in the above code sample. None of these match against the data that is retrieved by the code. Can you kindly help explain the delta ? 

[tmy_33.210_-95.720_2006_2015.csv](https://github.com/pvlib/pvlib-python/files/7780158/tmy_33.210_-95.720_2006_2015.csv)
[tmy_37.660_13.760_2006_2015.csv](https://github.com/pvlib/pvlib-python/files/7780161/tmy_37.660_13.760_2006_2015.csv)
[tmy_33.210_-95.720_2005_2014.csv](https://github.com/pvlib/pvlib-python/files/7780162/tmy_33.210_-95.720_2005_2014.csv)
[tmy_37.660_13.760_2005_2014.csv](https://github.com/pvlib/pvlib-python/files/7780163/tmy_37.660_13.760_2005_2014.csv)
[tmy_37.660_13.760_2007_2016.csv](https://github.com/pvlib/pvlib-python/files/7780165/tmy_37.660_13.760_2007_2016.csv)

AdamRJensen, 2021-12-27:
The reason that they differ is that the PVGIS website uses 10 year period for calculating the TMY, whereas the API defaults to using all years. This results in different months being used in the final TMY dataset.

From the [PVGIS API documentation](https://ec.europa.eu/jrc/en/PVGIS/docs/noninteractive):
[image]

The information is also included in the documentation of the pvlib function [get_pvgis_tmy](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.iotools.get_pvgis_tmy.html).

In order to obtain the same results as the website, it is necessary to specify `startyear` and `endyear`. E.g., for the first file you listed (tmy_33.210_-95.720_2006_2015.csv), the correct request would look like this:

    data, months_selected, inputs, meta = get_pvgis_tmy(33.21, -95.722, startyear=2006, endyear=2015)

Note that the specific months used in the TMY are given in the `months_selected` variable shown above. If you compare this variable with and without the `startyear`/`endyear` arguments you will see that the months used change.

UGuntupalli, 2021-12-27:
@AdamRJensen, 

Thank you for the clarification. PVsyst also offers a way to pull the PVGIS data (https://www.pvsyst.com/help/meteo_source_pvgis.htm). However, what is not clear at least reading this documentation is whether PVsyst is querying hourly data or monthly data. 

Is there a comparable form of getting the equivalent data as imported via Pvsyst using Pvlib ?

kanderso-nrel, 2021-12-27:
Hi @UGuntupalli, without knowing what exactly PVsyst is doing, I'm not sure we can advise how to replicate it in python.  Maybe someone else knows more about that PVsyst function.  That docs page you linked seems to say that it is no longer available in PVsyst anyway.

PR #1362 (2022-01-04): Add note concerning startyear/endyear to get_pvgis_tmy

 - [x] Closes #1360
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 ~~- [ ] Tests added~~
 ~~- [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 ~~- [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link [... cut]
closed 2022-01-04 by wholmgren