pvlib/pvlib-python #1363: Broken link in read_midc documentation

williamhobbs, 2021-12-29:
The link to Variable Style Rules (https://pvlib-python.readthedocs.io/en/latest/variables_style_rules.html) appears to be broken:

https://github.com/pvlib/pvlib-python/blob/b3c1b5e9fba606c3a89c052d07c9e500518bba5d/pvlib/iotools/midc.py#L155-L207

[image]

kanderso-nrel, 2021-12-29:
Thanks @williamhobbs, I didn't think to check for existing links when I moved those pages under `user_guide` in #1173.  Looking at the results of `$ git grep "pvlib-python\..*/en/"`, there are several links to update:

- `README.md:Please see the [Installation page](http://pvlib-python.readthedocs.io/en/stable/installation.html) of the documentation for complete instructions.`
- `docs/tutorials/irradiance.ipynb:    "http://pvlib-python.readthedocs.io/en/latest/clearsky.html\n",`
- ```pvlib/iotools/midc.py:    <https://pvlib-python.readthedocs.io/en/latest/variables_style_rules.html>`_.```
- `pvlib/modelchain.py:    See https://pvlib-python.readthedocs.io/en/stable/modelchain.html`

As an aside, is it desirable for any link like this to point to `latest`?  Maybe we should standardize to `stable` everywhere.

cwhanse, 2021-12-29:
> we should standardize to stable everywhere.

+1

PR #1365 (2022-02-15): Miscellaneous sphinx fixes
 - [x] Closes #1363 
 - [x] Closes #1300
 - [x] Closes #1377
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~Tests added~
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - ~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to t [... cut]
closed 2022-02-15 by kanderso-nrel