pvlib/pvlib-python #1364: Error when attempting to access NCEP weather model forecasts

StantonLanham, 2021-12-29:
**Describe the bug**
Getting a 404 error when attempting to retrieve weather model forecasts

> HTTPError: Error accessing https://thredds.ucar.edu/thredds/ncss/grib/NCEP/GFS/Global_0p5deg/Best/dataset.xml
Server Error (404: 404)

**To Reproduce**
Steps to reproduce the behavior:

> import pandas as pd
import datetime
from pvlib.forecast import GFS
latitude, longitude, tz = 32.2, -110.9, 'US/Arizona'
start = pd.Timestamp(datetime.date.today(), tz=tz)
end = start + pd.Timedelta(days=7)
irrad_vars = ['ghi', 'dni', 'dhi']
model = GFS()
raw_data = model.get_data(latitude, longitude, start, end)

**Versions:**
pvlib.__version__: 0.9.0
pandas.__version__: 1.3.2
python: 3.9.6
siphon.__version__: 0.9

**Additional context**
The above example is from the current documentation using the GFS model example. Also getting this error when attempting to use the HRRR model.

kahemker, 2021-12-30:
It seems that the Net CDF Subset service is down.  I have no idea who maintains this and where outages can be reported.

Start from this URL
https://thredds.ucar.edu/thredds/catalog/grib/NCEP/GFS/Global_0p5deg/latest.html?dataset=grib/NCEP/GFS/Global_0p5deg/GFS_Global_0p5deg_20211229_1800.grib2

Try to click "NetCDF Subset".   See the 404 error from the following link:  https://thredds.ucar.edu/thredds/ncss/grib/NCEP/GFS/Global_0p5deg/GFS_Global_0p5deg_20211229_1800.grib2/dataset.html

kahemker, 2021-12-30:
Sounds like Thredds is back online.  This issue can be closed

closed 2021-12-31 by StantonLanham