pvlib/pvlib-python #1367: Wrong Return in documentation of cloud_covr_to_transmittance_linear

JoLo90, 2022-01-05:
In the documentation of the **pvlib.forecast.NDFD.cloud_cover_to_transmittance_linear** function, it seems the documented return is the wrong one: It shows the return is the GHI instead of the transmittance.

[image]

Here is the function in pvlib.forecast that seems to confirm the case.

        """
        Convert cloud cover to atmospheric transmittance using a linear
        model.

        0% cloud cover returns offset.

        100% cloud cover returns 0.

        Parameters
        ----------
        cloud_cover : numeric
            Cloud cover in %.
        offset : numeric, default 0.75
            Determines the maximum transmittance.
        kwargs
            Not used.

        Returns
        -------
        ghi : numeric
            Estimated GHI.
        """
        transmittance = ((100.0 - cloud_cover) / 100.0) * offset

        return transmittance`

AdamRJensen, 2022-01-05:
Seems like it was a mistake when the [function](https://github.com/pvlib/pvlib-python/blame/master/pvlib/forecast.py#L512-L538) was added in #180.

Should probably be updated to something like this:

    Returns
    -------
    transmittance: numeric
        Estimated transmittance.

Perhaps it's worth denoting that it is a fraction and not in percentage like the cloud cover?

cwhanse, 2022-01-05:
> it's worth denoting that it is a fraction

Absolutely

AdamRJensen, 2022-01-05:
@JoLo90 If you wanted to make a pull request for this issue, that would be more than welcomed.

JoLo90, 2022-01-05:
@AdamRJensen I'll create a PR in the next days.

kanderso-nrel, 2022-01-16:
Closed by #1370, thanks @JoLo90!

closed 2022-01-16 by kanderso-nrel