pvlib/pvlib-python #1397: PVsystem.get_irradiance ignores surface tilt and surface azimuth

andrewmckayeaton, 2022-02-02:
**Describe the bug**
PVsystem.get_irradiance is a wrapper for irradiance.get_total_irradiance but when using PVsystem.get_total_irradiance the values of 'poa_global' do not change if the PV panel surface tilt and surface azimuth are changed whereas it seems to work as expected when using irradiance.get_total_irradiance.

On a related note, the method FixedMount.get_orientation is called with the arguments solar_zenith, solar_azimuth. The two arguments aren't necessary as the method just returns the attributes for surface tilt and azimuth.

**To Reproduce**
````
import pandas as pd
from pvlib import pvsystem, irradiance
from datetime import datetime

surface_tilt = 45
surface_azimuth = 180
module_capacity = 6000
data = {
  'zenith': 34,
  'azimuth': 180,
  'dni': 588,
  'dhi': 234,
  'ghi': 722
}

data = pd.DataFrame(data, index = [datetime.now()])

module_parameters = {
  'pdc0': module_capacity, 
  'gamma_pdc': -0.004, 
  'surface_tilt': surface_tilt, 
  'surface_azimuth': surface_azimuth
}

system = pvsystem.PVSystem(
  module_parameters=module_parameters
)

plane_of_array_irradiance = system.get_irradiance(
  data['zenith'], 
  data['azimuth'], 
  data['dni'], 
  data['ghi'], 
  data['dhi'], 
  model='perez', 
  model_perez='allsitescomposite1990'
)
plane_of_array_irradiance2 = irradiance.get_total_irradiance(
  surface_tilt, 
  surface_azimuth, 
  data['zenith'], 
  data['azimuth'], 
  data['dni'], 
  data['ghi'], 
  data['dhi']
)

print(f'Global PoA Irradiance (PVsystem calculation) = {plane_of_array_irradiance["poa_global"]}')
print(f'Global PoA Irradiance (Irradiance calculation) = {plane_of_array_irradiance2["poa_global"]}')
````

The two printed values will be different. Changing surface_tilt to 90 will change the calculation from the irradiance class to be the same as the PV system class calculation so I think PVsystem is always taking surface_tilt as 90.

**Expected behavior**
PVsystem.get_irradiance should give different values as surface_tilt and surface_azimuth change, in the same way that irradiance.get_total_irradiance does.

**Versions:**
pvlib == 0.9.0
pandas == 1.2.4

kanderso-nrel, 2022-02-02:
Hi @andrewmckayeaton, `surface_tilt` and `surface_azimuth` are direct parameters to `PVsystem`, not something to include in `module_parameters`.  See the discussion here: https://github.com/pvlib/pvlib-python/discussions/1326

You should also be sure that the comparison is apples-to-apples -- right now you're using the `perez` model in one case and the `isotropic` model (by default) in the other.  

> Changing surface_tilt to 90 will change the calculation from the irradiance class to be the same as the PV system class calculation so I think PVsystem is always taking surface_tilt as 90.

This is unexpected.  The default surface tilt is 0 for [PVSystem](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.PVSystem.html), so if you're supplying surface_tilt in the wrong place then it should default to 0.   Is it possible you meant 0 instead of 90? 

> On a related note, the method FixedMount.get_orientation is called with the arguments solar_zenith, solar_azimuth. The two arguments aren't necessary as the method just returns the attributes for surface tilt and azimuth.

It does seem a bit silly for the fixed-tilt case, but it might make more sense if you think about other types of Mount.  The `Array` and `PVSystem` classes are supposed to work for any mounting strategy, which means they don't know ahead of time what the Mount needs to calculate orientation.  `SingleAxisTrackerMount` needs the solar position inputs, so to have a common interface across all mounts (i.e., being able to call `mount.get_orientation(solar_zenith, solar_azimuth)` no matter what type of Mount it is), we do the same for `FixedMount`.

andrewmckayeaton, 2022-02-02:
Hi Kevin,

All of your suggestions worked, thanks for taking the time to explain everything. You were right, it was 0 I was seeing the same values for.

Thanks,
Andrew

closed 2022-02-02 by andrewmckayeaton

cwhanse, 2022-02-02:
> The default surface tilt is 0 for PVSystem

Perhaps `surface_tilt` should be required, or default to None...a thought.

wholmgren, 2022-02-02:
Or deprecated in favor of a required mount

andrewmckayeaton, 2022-02-02:
Just to add to the discussion, my mistake came from _feeling_ that surface_tilt and surface_azimuth were properties of the module since they definitely weren't properties of the inverter but if I understand correctly, they are properties of the mount. If that's the case then I would be in favour of @wholmgren's suggestion and requiring a mount_parameters dict as an argument to PVSystem.