pvlib/pvlib-python #1405: Klucher diffuse sky docstring error

adriesse, 2022-02-14:
The equation for F' should contain DHI rather than Id0.

**Versions:**
 - ``pvlib.__version__``: 0.9 and earlier

cwhanse, 2022-02-14:
I'm sure that Loutzenhiser's notation is `I_{d}` sky diffuse on a surface tilted at `d`, and `I_{d0}` is sky diffuse on a horizontal surface. But that's not helpful documentation here.

To do with this: add inline citations, punctuation, units in brackets, define `theta`, `theta_z`, and `beta`.

Maybe jettison the whole second paragraph that reiterates the inputs, and incorrectly adds `extraterrestrial irradiance`.

To discuss: remove "must be >=0", or change to "should be >=0". The code doesn't require this restriction, although the model certainly isn't valid outside that range. My view: model constraints on input (should be's) would be better placed in the description or in Notes, and code restrictions (must be's) in the parameter descriptions.

kanderso-nrel, 2022-03-16:
Seems like several of those items apply to the other sky diffuse model docstrings as well. 

Also, does it make sense for the Loutzenhiser reference to be listed first?  Seems like it ought to be a secondary reference if it's included at all.

cwhanse, 2022-03-16:
> Also, does it make sense for the Loutzenhiser reference to be listed first? Seems like it ought to be a secondary reference if it's included at all.

Although Loutzenhiser is a survey and comparison among models, it more clearly defines several models than do the source papers. That's why we chose it as the baseline reference.

PR #2192 (2024-09-26): Update irradiance.klucher docs
 - [X] Partially closes #2183, closes #1405
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [X] New  [... cut]
closed 2024-09-26 by kandersolar