pvlib/pvlib-python #1409: DC model unexpectedly fills NaNs with 0

harakiricode, 2022-02-18:
I would like to report a behaviour in the dc models of pvlib, which was very unintuitive to me. Since it has been coded very explicitly though, I assume the behaviour has been considered.

I am on version 0.6.1 of pvlib, but the issue occurs on the current version, too. Using, for example the desoto dc model, NaNs are filled with 0 after calling scale_voltage_to_current_power. ([v0.6.1](https://github.com/pvlib/pvlib-python/blob/v0.6.1/pvlib/modelchain.py#L438l) and [current master](https://github.com/pvlib/pvlib-python/blob/master/pvlib/modelchain.py#L778)). In my specific case, this means, that after having passed NaNs to the model chain in my weather dataframe, I receive valid numbers in the model chain result. I would have either expected the afflicted time steps to be NaN, a warning or an error.

Is this behaviour by the dc model - and as a result by the entire model chain - intended?

cwhanse, 2022-02-18:
You raise a fair point. That has been the behavior since the original `ModelChain.singlediode` method in v0.4.0. I can't say exactly why; my guess is so that `ModelChain.result.dc.sum()` returns a valid value.

wholmgren, 2022-02-23:
I agree that behavior is very bad and inconsistent with the general pvlib policy of "nan in, nan out". I don't remember, but if I had to guess, I may have added it as a very lazy way of handling missing nighttime values.

A pull request would be welcome.

matsuobasho, 2024-10-16:
@cwhanse I'm trying to find a more challenging issue to tackle that will be most useful to users (so looked at the 'help wanted' ones).  I'm not using pvlib every day so don't have such a good gauge.  This is an older one, would like some guidance whether this will be useful to resolve now.  If not, can you point to another one that's active and is a pain point right now?  Perhaps this one https://github.com/pvlib/pvlib-python/issues/2245

cwhanse, 2024-10-20:
@matsuobasho either #1409 or #2245 would be welcome. I think @2245 is easier. Here, we recorded the issue but didn't investigate further. my fear is numerous little edits needed to tests.

#1711 would also be nice to close.