pvlib/pvlib-python #1411: It's a small inelegance that specifying `aoi_model` is necessary for a ModelChain that only uses `run_model_from_effective_irradiance`.  I don't see a way around it, so I don't think there's anything to be done in this PR, just thinking out loud about the ModelChain design.

cwhanse, 2022-02-24:
It's a small inelegance that specifying `aoi_model` is necessary for a ModelChain that only uses `run_model_from_effective_irradiance`.  I don't see a way around it, so I don't think there's anything to be done in this PR, just thinking out loud about the ModelChain design.

_Originally posted by @kanderso-nrel in https://github.com/pvlib/pvlib-python/pull/1394#r798706567_

wholmgren, 2022-02-24:
> It would be nice if the method runs without requiring these attributes.

Isn't the problem that the method must be specified when the `ModelChain` object is instantiated? If so, a better design might be different `ModelChain` classes such as `ModelChain3Component`, and `ModelChainEffectiveIrradiance`. Then each would have its own `run_model` rather than `run_model_.*`. Of course that's a pretty substantial API change and may or may not be worth it. Setting different defaults and raising errors at runtime if needed would be a relatively small change, but is less consistent with one of the original goals of `ModelChain` (if you can build it, you can run it).

cwhanse, 2022-02-24:
> Isn't the problem that the method must be specified when the ModelChain object is instantiated?

Yes, and please feel at liberty to edit the issue title and description.