pvlib/pvlib-python #1412: Improve ``get_cams`` documentation

AdamRJensen, 2022-02-24:
**Describe the bug**
The geographical coverage of CAMS Radiation retrieved with the [``get_cams``](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.iotools.get_cams.html#pvlib.iotools.get_cams) function is not exactly ±66 degrees in latitude and longitude. Therefore, to clarify this, the word "approximate" should be added to the function documentation [here](https://github.com/pvlib/pvlib-python/blob/63c36db473ce9d7406bfb90526c37783515716c2/pvlib/iotools/sodapro.py#L59).

Also, the description has a typo - "Wordwide" should be corrected to "worldwide".

**Screenshots**
The actual CAMS Radiation coverage:
[image]

AdamRJensen, 2022-02-27:
@soma2000-lang That would be much appreciated! :)

soma2000-lang, 2022-03-01:
Ok working on that

PR #1416 (2022-03-03): Improve get_cams documentation

 - [x] Closes #1412
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.
closed 2022-03-03 by cwhanse