pvlib/pvlib-python #1415: Changes to PVGIS in version 5.2

AdamRJensen, 2022-02-27:
**Describe the bug**
PVGIS is releasing version 5.2 on March 1st, 2022. In the new version, the irradiance databases CMSAF and COSMO have been removed. Additionally, the ERA5 coverage has been extended to cover almost the entire globe and a new database, SARAH-2, has been added.

The new default databases can be seen here:
[image]

**Additional context**
There is nothing that indicates any changes to the API, so the pvlib.iootools functions ``get_pvgis_tmy`` and ``get_pvgis_hourly`` should continue to work.

However, some of the test files used in `test_pvgis.py` are from the CMSAF database. Since those tests seem to be either based on local files or mock tests, I do not expect test failures. However, it would probably make sense to update the tests regardless.

cwhanse, 2022-02-28:
@AdamRJensen do you know if the content of CMSAF and COSMO are migrating to the new database? Sorry that I'm not familiar with PVGIS.

cwhanse, 2022-02-28:
> @AdamRJensen do you know if the content of CMSAF and COSMO is migrating to the new database? Sorry that I'm not familiar with PVGIS.

I missed the link to the documentation in my quick read of your issue, the documentation answers my question. I agree it would make sense to replace the test files or (more simply) to add a note to the test code identifying the data source.

AdamRJensen, 2022-03-01:
In order to use the new version 5.2 (which is still in beta), the URL needs to be updated, e.g.,:

    res = pvlib.iotools.get_pvgis_hourly(
        latitude=28.965734,
        longitude=117.147418,
        raddatabase='PVGIS-ERA5',
        url='https://re.jrc.ec.europa.eu/api/v5_2/',
    )

I suspect this will only be necessary temporarily until the databases are considered stable and migrated to the default URL. The retired databases CMSAF and COSMO remain available through the default URL for now, though they will be removed in the future.

kanderso-nrel, 2022-03-21:
Probably a good idea to scrub mentions of the retired databases from the docstring as well:

https://github.com/pvlib/pvlib-python/blob/c2431832040f585969bbd1868f6acdd6ddb4e0a5/pvlib/iotools/pvgis.py#L140-L144

@AdamRJensen thoughts on tagging this issue for 0.9.1?

AdamRJensen, 2022-03-21:
@kanderso-nrel what's the current timeline for 0.9.1? I'll be out of the office until April 5th fyi.

kanderso-nrel, 2022-03-22:
> what's the current timeline for 0.9.1

Don't think there really is a timeline.  Next couple weeks maybe?  Depends on how long it takes us to finish or defer the remaining things in #1424...

kanderso-nrel, 2022-07-22:
@AdamRJensen should we keep this open until we update the tests and docstrings?

AdamRJensen, 2022-07-22:
@kanderso-nrel Sorry I got ahead of myself

PR #1502 (2022-08-15): Update PVGIS docs and test to comply with PVGIS version 5.2

 - [x] Closes #1415
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~~
 - ~~[ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes [... cut]
closed 2022-08-15 by AdamRJensen