pvlib/pvlib-python #1417: Forecasting page in User's guide

cwhanse, 2022-03-02:
The [Forecasting](https://pvlib-python.readthedocs.io/en/stable/forecasts.html) page has at least two issues:

- the "View on Github" link is broken
- the example code refers to a pvlib location instance that is not defined. It is referenced in a block of code that isn't run, so doesn't generate an error. Including this code can be [confusing](https://groups.google.com/g/pvlib-python/c/nkSokU0ORfE/m/sono772cBwAJ) to new users.

kanderso-nrel, 2022-03-02:
Looks like the NDFD section has some broken code too.  

> the "View on Github" link is broken

In fact most of the top-level pages in `stable` have broken "view on github" links.  This is because the 0.9.0 docs still point to the old repository locations from before #1173 moved those pages to the new `user_guide` subdirectory.  I don't think there's anything to do about it until we release 0.9.1 and the docs get rebuilt with the new repository locations.

cwhanse, 2022-03-02:
On second look, the block of code referencing the `location` instance isn't execute, it's there to show how ghi is obtained from cloud cover. Maybe we can display that information differently to avoid confusing new users, who copy the code and try to run it.

soma2000-lang, 2022-03-03:
@cwhanse  Can I work on this??

cwhanse, 2022-03-03:
@soma2000-lang certainly. My idea is to replace these lines with math (for the last three lines) and provide a reference to the [cloud_cover_to_irradiance_clearsky_scaling](https://github.com/pvlib/pvlib-python/blob/df1c56e92d91701319adad4fe5cf98837edad0fb/pvlib/forecast.py#L466) function

E.g., 

GHI = [offset + (1 - offset) x (1 - cc) ] x GHI_{clearsky}

where offset (% of clearsky GHI) is the minimum GHI under complete cloud cover, and cc is the cloud cover (fraction of sky dome).

DNI is computed using the DISC model <link to function here>, and DHI is computing as

<equation for DHI>

```
.. code-block:: python
    solpos = location.get_solarposition(cloud_cover.index)
    cs = location.get_clearsky(cloud_cover.index, model='ineichen')
    # offset and cloud cover in decimal units here
    # larson et. al. use offset = 0.35
    ghi = (offset + (1 - offset) * (1 - cloud_cover)) * ghi_clear
    dni = disc(ghi, solpos['zenith'], cloud_cover.index)['dni']
    dhi = ghi - dni * np.cos(np.radians(solpos['zenith']))
```

PR #1766 (2023-06-23): Remove `pvlib.forecast`
 - [x] Closes #1596 
 - [x] Closes #1417
 - [x] Closes #918
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link t [... cut]
closed 2023-06-23 by kandersolar