pvlib/pvlib-python #1423: pvlib.forecast has stopped working

DavidLloydNGP, 2022-03-10:
Hi,

The example code for importing a forecast from GFS has stopped working for me.
```

from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP
import pandas as pd
import datetime

latitude, longitude, tz = 32.2, -110.9, 'US/Arizona'

start = pd.Timestamp(datetime.date.today(), tz=tz)

end = start + pd.Timedelta(days=7)

irrad_vars = ['ghi', 'dni', 'dhi']

model = GFS()

raw_data = model.get_data(latitude, longitude, start, end)

print(raw_data.head())
```
returns an error now:
```
lib\xml\etree\ElementTree.py", line 1321, in XML
    return parser.close()

  File "<string>", line unknown
ParseError: no element found: line 1, column 0
```

python version is 3.8 and numpy version is 1.21.2.

Has anyone else encountered this problem?

I was able to read in forecasted data without problems previously but I don't know what's changed between then and now.

Thanks,
David

cwhanse, 2022-03-10:
@DavidLloydNGP I've seen that error when an empty file is returned by the THREDDS server. In this case, it is [this line](https://github.com/pvlib/pvlib-python/blob/df1c56e92d91701319adad4fe5cf98837edad0fb/pvlib/forecast.py#L139). A few lines earlier, the `model_url` is constructed: "https://thredds.ucar.edu/thredds/catalog/grib/NCEP/GFS/Global_0p5deg/catalog.xml" Pasting that URL in a browser returns an empty page.

This appears to be a THREDDS issue. Perhaps pvlib could return a more informative error somehow.

DavidLloydNGP, 2022-03-11:
Thanks @cwhanse for that. Subsequent to my post, I had even tried running the code in a new virtual environment, hoping it was a problem with another library but I received the same error. I think this supports your analysis that something is amiss further upstream with THREDDS.

I managed to 'solve' this problem by bypassing pvlib and using siphon directly to access GFS data. For instance, to download the wind data for the next 24 hours for Greenwich, London:

```
# adapted from
#https://unidata.github.io/python-training/workshop/Bonus/downloading-gfs-with-siphon/

from siphon.catalog import TDSCatalog
import datetime
import pytz

# list of relevant wind variables
var_list = ['u-component_of_wind_isobaric','v-component_of_wind_isobaric',\
            'Wind_speed_gust_surface']

# query 0.5 degree model, same as pvlib
best_gfs = TDSCatalog('http://thredds.ucar.edu/thredds/catalog/grib/NCEP/GFS/'
                      'Global_0p5deg/catalog.xml?dataset=grib/NCEP/GFS/Global_0p5deg/Best')
best_ds = list(best_gfs.datasets.values())[0]
ncss = best_ds.subset()

query = ncss.query()
    
# single geographic location
query.lonlat_point(lat = 0, lon = 0)

# read in data for next 24 hours
timezone = pytz.timezone("Europe/London")
T0 = datetime.datetime.now().astimezone(timezone)
query.time_range(T0, T0 + datetime.timedelta(days = 1))

query.accept('netcdf4')

# loop over wind variables
for var in var_list:
    query.variables(var)
    
query.vertical_level(100000)

dataset = ncss.get_data(query)

surface_wind_gust = dataset.variables['Wind_speed_gust_surface'][0].data
u_isobaric = dataset.variables['u-component_of_wind_isobaric'][0].data
v_isobaric = dataset.variables['v-component_of_wind_isobaric'][0].data
```

I have done a few non-exhaustive checks and the data obtained this way seems identical to data I had previously downloaded using pvlib. Given the pvlib forecast function always came with a 'highly experimental' warning anyway, I think I am going to stick with the siphon approach, even it requires a little more coding on my part to replicate the ease of pvlib.

kanderso-nrel, 2022-03-11:
FWIW the example code at the top of this thread seems to work again.  Still worth considering a nicer error message though.

cwhanse, 2022-03-16:
THREDDS appears to be working (at least today it is). Closing in anticipation on #1426

closed 2022-03-16 by cwhanse