pvlib/pvlib-python #1425: Change PSM3 map_variables default value

kanderso-nrel, 2022-03-14:
#1374 did what we should probably be doing with most deprecations: mention the specific version when the deprecated behavior will change.  

https://github.com/pvlib/pvlib-python/blob/1893b20a7b755004f561037161c242db24e2870c/pvlib/iotools/psm3.py#L334-L340

Tagging this in the 0.11.0 milestone so we don't forget it.

mikofski, 2022-03-15:
isn't this part of some Sphinx directive somewhere? I always this markup that says like
> deprecated since v0.X.Y 

see: https://deprecated.readthedocs.io/en/latest/sphinx_deco.html
```
from deprecated.sphinx import deprecated

@deprecated(
    reason="""Since Python 3.4, you can use the standard function :func:`statistics.mean`.""",
    version="2.5.0",
)
def myfun(args):
    """
    Foobar.

    Blah.
    """
    return args
```

kanderso-nrel, 2022-03-15:
Well the `pvlib._deprecation.deprecated` has `since` and `removal` parameters and inserts that `.. deprecated::` directive into the docstring too.  But it inserts the directive at the top of the docstring so it applies to the entire function, not a single parameter.

I'm not sure I follow the suggestion though.  Are you proposing we make a note in the docstring's description of `map_variables` that the default value will change in the future?

mikofski, 2022-03-15:
Sorry, I didn't think it through very well. I guess, this warning for `map_variables` seems like a maintenance burden, because we'll the deprecation version is hard-coded, so we need a follow up issue to update the warning message after v0.11 I think. It would be nice if there was a boilerplate way to have this somewhat automated, but probably not. It's not that big a deal

PR #2094 (2024-06-19): Set `map_variables=True` by default in PSM3 iotools functions
 - [x] Closes #1425
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link t [... cut]
closed 2024-06-19 by kandersolar