pvlib/pvlib-python #1428: Value error when reading data['Time (HH:MM)']

Nitzan-hez, 2022-03-18:
**Describe the bug**
On tmy.py, I got value error due wrong index if of string parenthesis - related to mismatch between required time format XX:00 to the time format saved with my csv file (X:00)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Enable tmy using bifacial.py package with the default csv data came with the package
2. Failure on line 184:
    shifted_hour = data['Time (HH:MM)'].str[:2].astype(int)%24
""'
ValueError: invalid literal for int() with base 10: '1:'
"""
3. occur due expected XX:00 format but real data was saved as X:00 format- for example instead of 01:00 data was saved as 1:00.

**Suggested Solution**
Padding the time string with zero in case length is 4 instead of 5 solved the issue, see new line below:
(184)   shifted_hour = data['Time (HH:MM)'].str.pad(5,fillchar='0').str[:2].astype(int) % 24
instead of
(184)   shifted_hour = data['Time (HH:MM)'].str[:2].astype(int) % 24

kanderso-nrel, 2022-03-18:
> with the default csv data came with the package

Is this a file in the [pvlib/data](https://github.com/pvlib/pvlib-python/tree/master/pvlib/data) folder? Can you provide the name of the specific file you are trying to read?

I think I lean towards not calling this a bug: Table 1-3 in the [TMY3 User's Manual](https://www.nrel.gov/docs/fy08osti/43156.pdf) (listed in the references section for [read_tmy3](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.iotools.read_tmy3.html)) says the time column is formatted as `HH:MM`, so a file with `H:MM` or `(H:MM)` is not in the TMY3 format, and therefore it's not a bug if `read_tmy3` can't read it.

Nitzan-hez, 2022-03-18:
> Is this a file in the [pvlib/data](https://github.com/pvlib/pvlib-python/tree/master/pvlib/data) folder? Can you provide the name of the specific file you are trying to read?
>
> I think I lean towards [...]

closed 2022-03-18 by Nitzan-hez