pvlib/pvlib-python #1430: pvlib.solarposition.spa_python has unused and undocumented **kwargs

kanderso-nrel, 2022-03-23:
`pvlib.solarposition.spa_python` has `**kwargs` in the signature, but it doesn't mention it in the docstring, and it doesn't do anything with it in the function body.

https://github.com/pvlib/pvlib-python/blob/c2431832040f585969bbd1868f6acdd6ddb4e0a5/pvlib/solarposition.py#L276-L278

This is bad because it silently ignores misspelled optional parameters, for example: `pvlib.solarposition.spa_python(times, 40, -80, atlitude=100)`.  

---

We could just deprecate and remove `**kwargs` like #1053:

https://github.com/pvlib/pvlib-python/blob/a6667a390d048d4301f73acef3267596dc1fafcd/pvlib/location.py#L83-L87

Or we could keep `**kwargs` and pass them through to `spa.solar_position()`, thereby exposing the `sst` and `esd` parameters.  Though if we want to expose those, I'd prefer just exposing them directly instead of hiding them in `**kwargs`.

This could be a good "easy first issue" once we decide what the fix should be.

wholmgren, 2022-03-23:
I'd support removing without deprecation. Only impact on users will be to reveal bugs in their code.

Naman-Priyadarshi, 2022-03-28:
I'll work on the issue

kanderso-nrel, 2022-03-30:
Looks like `Location.get_clearsky()` should be more careful about what it passes through to the solar position functions; there's no reason to pass turbidity, airmass etc through to any solar position model right?  Maybe `_build_kwargs` is the solution here, or just using `kwargs.pop()` like `Location.get_clearsky()` does for the clearsky portion.

cwhanse, 2022-03-30:
I agree, no reason that any solar position function needs turbidity. What do you think about handling the `**kwargs` in `pvlib.solarposition.get_solarposition` instead of in the `Location` methods?

kanderso-nrel, 2022-04-04:
> What do you think about handling the **kwargs in pvlib.solarposition.get_solarposition instead of in the Location methods?

If the handling raises a TypeError when something like `linke_turbidity` is passed in, I don't think it avoids the problem of needing to change how `Location.get_clearsky()` passes `kwargs` around.  And if it doesn't raise an error for inappropriate inputs, I think it's just shifting the problem of "silently ignoring misspelled optional parameters" to a different, and probably more commonly used, function.

On the other hand, I don't like trying to subset `**kwargs` in `Location.get_clearsky()` either since having `location.py` keep track of the many and various solar position parameters is not DRY.

Noting that it currently doesn't work to specify solar position kwargs in `Location.get_clearsky()` anyway (`TypeError: ineichen() got an unexpected keyword argument 'temperature'`), maybe the easiest way forward here is to just stop using `**kwargs` for the solar position part of `Location.get_clearsky()`.  No functionality is lost, and people that want to customize it can still pass in their own solar position. 

---

Semi-related: `pvlib.irradiance.get_total_irradiance` also has unused `**kwargs`; might as well fix that along with `spa_python`.

closed 2022-04-06 by kanderso-nrel