pvlib/pvlib-python #1439: pvlib-python-asv.conf

roger-lcc, 2022-04-09:
**Describe the bug**
In asv.conf, the virsion of numba is conflictde with the virsion of numpy,so when run this code `asv check`,there is a bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Validate and build environments (may take a couple minutes to run): `asv check`
2. See error

**Expected behavior**
modify the asv.conf 

**Screenshots**
[image]

**Versions:**
 - ``pvlib.__version__``: 
 - ``pandas.__version__``: 
 - python:conda python 3.9

**Additional context**
And i change the virsion  0.36.1 of numba to the virsion 0.40.0 of numba ,it works.

Naman-Priyadarshi, 2022-04-11:
I tried to recreate the same behavior to get the error but my `asv check` ran successfully. When I published my result the graphs were empty. So, I think we should remove the python 3.6 environment as it has reached [EOL](https://devguide.python.org/#status-of-python-branches) and add another environment for python 3.9 or 3.10 and update all the versions to the latest for numpy, pandas, scipy, h5py, ephem, and numba.

roger-lcc, 2022-04-11:
I think the result of each benchmark is just one commit result because we just ran `asv run`.If we click into the graphs, we will see the results of benchmarks.
[image]
And I think if we run `asv run v0.9.0`——the virsion of pvlib, and we could get the results of all the commits.  But the process of the benchmarks is too long, so I have not fininshed it.

roger-lcc, 2022-04-11:
And I tried again the default asv.conf, it still had the problem.And the test of benchmarks is upon the conda virtual environment which asv created itself.So I think maybe the problem is still about asv.conf.

closed 2022-04-12 by roger-lcc