pvlib/pvlib-python #1442: Temperature model parameter translation

adriesse, 2022-04-13:
**Is your feature request related to a problem? Please describe.**
The simple temperature models in pvlib are all very similar but their parameters are not interchangeable.

**Describe the solution you'd like**
This feature will provide a means to translate parameters between models.

**Additional context**
To be presented at IEEE PVSC 2022 and documented in a corresponding paper.

mikofski, 2022-05-29:
This is an interesting idea. Does it only refer to [Faiman](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.temperature.faiman.html) and [PVsyst](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.temperature.pvsyst_cell.html)? Or also Fuentes or SAPM?

perhaps a write up in a new temperature section of the [user guide](https://pvlib-python.readthedocs.io/en/stable/user_guide/index.html), on the PVPMC website, or in the pvlib GH wiki would be useful to describe in more detail how these models are mathematically and scientifically similar or different would be helpful in explaining why we need a function or class to interpret them and use them interchangeably?

adriesse, 2022-05-29:
I updated the current scope in the opening description: Faiman, SAPM, PVsyst & NOCT SAM.  The upcoming IEEE PVSC paper should answer a lot of your questions and we can see after the event how much of that content needs repeating in other forms.

kanderso-nrel, 2022-09-09:
@adriesse I've been lead to understand that this feature and #1523 are the first of several planned additions to pvlib.  What do you think about opening (preliminary) issues for those future additions now?  For one it's just interesting to see what's coming down the pipe, but it might also be useful context for other additions in the meantime.

adriesse, 2022-09-09:
That's a good question but hard to answer.  We're doing model development and the output can go into functions, notebooks, reports and papers and can include data files.  I could try to open PR's earlier, but that's probably not a full solution for coordinated development.

PR #1463 (2022-09-13): Temperature model parameter translation code

 - [x] Closes #1442 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub  [... cut]
closed 2022-09-13 by kanderso-nrel