pvlib/pvlib-python #1460: pvlib.solarposition.sun_rise_set_transit_spa returning wrong sunrise time

Matthew-Mullin, 2022-05-20:
**Describe the bug**
The function pvlib.solarposition.sun_rise_set_transit_spa is outputting what seems to be the wrong sunrise time for all the parameters I'm using.

**To Reproduce**
Steps to reproduce the behavior:

```
import numpy as np
import pandas as pd
import datetime
from pvlib import solarposition as sp

def get_sunrise_hour(month, day, year, lat, lon, timezone):
    times = pd.DatetimeIndex(data=[datetime.date(year,month,day)],tz=timezone)
    sunrise_df = sp.sun_rise_set_transit_spa(times=times,latitude=lat,longitude=lon,delta_t=None)
    return (sunrise_df['sunrise'].array[0].hour + sunrise_df['sunrise'].array[0].minute/60)

month=3; day=20; year=2022; lat=34.867; lon=-116.783; timezone=-8;

print(get_sunrise_hour(month,day,year,lat,lon,timezone))

```

Prints out 13.833 (which is clearly wrong)

**Expected behavior**
expects to print something like 5,6,or 7 for 5,6,7AM

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``:  0.9.0
 - ``pandas.__version__``: 1.4.2
 - python: 3.9.7

**Additional context**
Add any other context about the problem here.

kanderso-nrel, 2022-05-20:
Hi @Matthew-Mullin, I think the issue is how you are specifying the timezone.  It looks like pandas interprets a numeric timezone not as a UTC offset in hours like you probably expected, but rather a UTC offset in seconds.  Compare the offsets (`-00:00:08` vs `-08:00`):

```python
In [16]: pd.DatetimeIndex([datetime.date(2022, 3, 20)], tz=-8)
Out[16]: DatetimeIndex(['2022-03-20 00:00:00-00:00:08'], dtype='datetime64[ns, pytz.FixedOffset(0)]', freq=None)

In [17]: pd.DatetimeIndex([datetime.date(2022, 3, 20)], tz='Etc/GMT+8')
Out[17]: DatetimeIndex(['2022-03-20 00:00:00-08:00'], dtype='datetime64[ns, Etc/GMT+8]', freq=None)
```

Note: in case you're not aware, the sign convention in `Etc/GMT+8` is the opposite of what you might expect.  See https://pvlib-python.readthedocs.io/en/latest/user_guide/timetimezones.html#fixed-offsets

Matthew-Mullin, 2022-05-20:
My bad, I didn't realize the function returned UTC. Here's the code for it to return local time:

```
def get_sunrise_hour(month, day, year, lat, lon, timezone):
    times = pd.DatetimeIndex(data=[datetime.date(year,month,day)],tz=timezone)
    sunrise_df = solarposition.sun_rise_set_transit_spa(times=times,latitude=lat,longitude=lon,delta_t=None)
    from_zone = tz.gettz('UTC')
    to_zone = tz.tzoffset('UTC', (3600*timezone))
    sunrise = sunrise_df['sunrise'].array[0].replace(tzinfo=from_zone)
    sunrise_local = sunrise.astimezone(to_zone)
    return (sunrise_local.hour + sunrise_local.minute/60)
```

closed 2022-05-20 by Matthew-Mullin