pvlib/pvlib-python #1462: NOCT_SAM may not handle effective irradiance correctly

adriesse, 2022-05-24:
**Describe the problem**
The noct_sam temperature model has an `effective_irradiance` input, which would normally have a value less than `poa_global`, mainly due to reflection losses.  This is implemented internally as an adjustment to the `transmittance_absorptance` parameter.  However, this  adjustment produces some questionable results.

**Code fragment to demonstrate the problem**
```
from pvlib.temperature import noct_sam

gpoa = 800
geff = gpoa * 0.75

wind_speed_10m = 1.0 / 0.51

NOCT = 41.6

# open circuit and max power module conditions for reuse
module_oc =dict(noct=NOCT, module_efficiency=0.00)
module_mp =dict(noct=NOCT, module_efficiency=0.20)

weather0 = dict(poa_global=gpoa, 
                temp_air=20, 
                wind_speed=wind_speed_10m,
                )
print()
print('mod temp under gpoa')
print('oc:', noct_sam(**weather0, **module_oc))
print('mp:', noct_sam(**weather0, **module_mp))

weather1 = dict(poa_global=geff, 
                temp_air=20, 
                wind_speed=wind_speed_10m,
                )
print()
print('mod temp under geff (what I would expect to be correct)')
print('oc:', noct_sam(**weather1, **module_oc))
print('mp:', noct_sam(**weather1, **module_mp))

# use the effective_irradiance input
weather2 = dict(poa_global=gpoa, 
                temp_air=20, 
                wind_speed=wind_speed_10m,
                effective_irradiance=geff,
                )
print()
print('mod temp under gpoa using separate geff input (different/incorrect)')
print('oc:', noct_sam(**weather2, **module_oc))
print('mp:', noct_sam(**weather2, **module_mp))

# replicate the tau_alpha adjustment that is inside noct_sam
weather3 = dict(poa_global=gpoa, 
                temp_air=20, 
                wind_speed=wind_speed_10m,
                transmittance_absorptance=0.9 * geff / gpoa,
                )
print()
print('mod temp under gpoa with tau_alpha adjusted externally (same problem)')
print('oc:', noct_sam(**weather3, **module_oc))
print('mp:', noct_sam(**weather3, **module_mp))

```
**Code fragment output**
```
mod temp under gpoa
oc: 41.6
mp: 36.8

mod temp under geff (what I would expect to be correct)
oc: 36.2
mp: 32.6

mod temp under gpoa using separate geff input (different/incorrect)
oc: 41.6
mp: 35.2

mod temp under gpoa with tau_alpha adjusted externally (same problem)
oc: 41.6
mp: 35.2
```
**Comments on output**
Note that under open circuit conditions the module temperature remains at 41.6 in the third and fourth cases despite the reduction in irradiance, which is not logical. The biggest surprise comes when you set `geff` to small values, like 20% of `gpoa` because the module temperature then goes below ambient.

**Additional context**
The `noct_sam` function in pvlib appears to correspond very closely to the code in SAM itself, so this problem would exist there too.  But it is much easier to demonstrate it here.

**Solution**
Reflected irradiance does not contribute to heating or electricity generation, so a single irradiance input for this thermal model is enough.

cwhanse, 2024-11-11:
Update on this isseu: SAM code has changed to use `effective_irradiance` instead of `poa_global` for the `noct_sam` temperature model https://github.com/NREL/ssc/pull/1233. SAM considers the change to be correcting a bug.

I think pvlib should correct its code, with liberal in-line comments to document that the correction deviates from the published reference.

adriesse, 2024-11-11:
I support a correction (of course).  I haven't had time to look closely at the change in SAM though. Did they write anything about impact on results?  Since it has been this way for so long we might consider offering both behaviors.

pratham15541, 2025-02-14:
I think we would need to modify the noct_sam function to handle the effective_irradiance input correctly or remove the effective_irradiance input altogether, using a single irradiance input for the thermal model.

cwhanse, 2025-02-14:
@pratham15541 can you clarify your comment? `noct_sam` doesn't current have `effective_irradiance` as input, and this issue's consensus is that `effective_irradiance` should replace `poa_global`.

Chirag3841, 2025-11-28:
@cwhanse I would like to work on this issue if any other person is not doing that .