pvlib/pvlib-python #1465: Retrieve data from FGS with specific time range via pvlib

najiya-12, 2022-05-28:
I was trying to get forecast data within a specific time range.  I am not sure what is the problem with data retrieval as stated here!!

`from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP
latitude, longitude, tz = 39.742, -105.18, 'UTC'
start = pd.Timestamp(datetime.datetime(2022, 4,26), tz=tz)
end = pd.Timestamp(datetime.datetime(2022,4, 26), tz=tz)
irrad_vars = ['ghi', 'dni', 'dhi']
model1= GFS()
dataGFS = model1.get_data(latitude, longitude, start, end)
print(dataGFS.head())
`

HTTPError: Error accessing https://thredds.ucar.edu/thredds/ncss/grib/NCEP/GFS/Global_0p5deg/Best?var=Medium_cloud_cover_middle_cloud_Mixed_intervals_Average&var=Total_cloud_cover_convective_cloud&var=Low_cloud_cover_low_cloud_Mixed_intervals_Average&var=High_cloud_cover_high_cloud_Mixed_intervals_Average&var=Temperature_surface&var=v-component_of_wind_isobaric&var=Wind_speed_gust_surface&var=u-component_of_wind_isobaric&var=Total_cloud_cover_boundary_layer_cloud_Mixed_intervals_Average&var=Total_cloud_cover_entire_atmosphere_Mixed_intervals_Average&var=Downward_Short-Wave_Radiation_Flux_surface_Mixed_intervals_Average&time_start=2022-04-26T00%3A00%3A00%2B00%3A00&time_end=2022-04-26T00%3A00%3A00%2B00%3A00&longitude=-105.18&latitude=39.742&vertCoord=100000&accept=netcdf
Server Error (400: Requested time range 2022-04-26T00:00:00Z - 2022-04-26T00:00:00Z does not intersect actual time range 2022-04-28T00:00:00Z - 2022-06-12T18:00:00Z)

wholmgren, 2022-05-31:
Hi @najiya-12 it is not possible to use pvlib to obtain forecast data outside of the short archive that is available on the Unidata server. For more, please search the pvlib python google group for previous discussions on this topic. Please also note that the `forecast.py` module is deprecated and will be removed soon.

closed 2022-05-31 by wholmgren

najiya-12, 2022-06-01:
Thank you @wholmgren! any suggestions for an automated way of downloading the same data