pvlib/pvlib-python #1481: Set leap_day=True as default in pvlib.iotools.get_psm3()

williamhobbs, 2022-06-22:
**Is your feature request related to a problem? Please describe.**
Several functions require equal timesteps with no gaps (e.g., `pvlib.temperature.prilliman` and `detect_clearsky`). Having weather data that excludes leap day will cause an error. Specifically, using `pvlib.iotools.get_psm3` for a leap year with the default `leap_day=True` would be a common reason for encountering this error.

**Describe the solution you'd like**
As suggested by @adriesse here https://github.com/pvlib/pvlib-python/issues/1476#issuecomment-1163130675, perhaps the default should be changed to `leap_day=True`?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Related to https://github.com/pvlib/pvlib-python/issues/1476. 

On the topic of "is this a good idea", some comments from issue 1476:
> "I wonder if there are any compelling reasons to keep it as False, other than the fact that other tools (e.g., NREL SAM) can't handle leap day." 

(by me)

> "I suspect `leap_day=False` is rooted in matching the PSM3 API's decision to set it False by default." 

(by @kanderso-nrel)

wholmgren, 2022-06-22:
I want to make sure I understand the API... PSM3 ignores ``leap_day`` for tmy/tgy requests and always returns a year with 365 days? But defaults to skipping the leap day when requesting, say, 5 minute data for 2020? If I understand correctly, I agree that the default should change.

williamhobbs, 2022-06-22:
In regards to the _NREL NSRDB API_, there is no `leap_day` parameter for [PSM3 typical years](https://developer.nrel.gov/docs/solar/nsrdb/psm3-tmy-download/) like there is for [hourly PSM3](https://developer.nrel.gov/docs/solar/nsrdb/psm3-download/) and [5min PSM3](https://developer.nrel.gov/docs/solar/nsrdb/psm3-5min-download/). I think it is safe to assume that the PSM3 TMY API will never return a leap day, even if February for the given TXY dataset happened to come from a leap year. But I've never seen that documented. 

For `pvlib.iotools.get_psm3`, I'll defer to someone else to confirm, but I think you are correct on both points.

adriesse, 2022-06-22:
A potential alternative would be not to have a default behavior in order to force users to make a conscious choice.

AdamRJensen, 2022-06-29:
I do not know of any other service that excludes leap days when requesting normal (non TMY) time-series data; hence, I think changing the ``get_psm3`` to include leap days by default would be in line with the remaining iotools and the general expectations of users. I also prefer this to not having a default choice. After all real-world systems do indeed observe leap days!

kanderso-nrel, 2022-07-19:
Any thoughts on whether this deserves a deprecation cycle like we've done for `map_variables` in #1374?  I could go either way.

AdamRJensen, 2022-07-22:
I'm ok with no deprecation cycle. I suppose then it should first be implemented in 0.10 (change the milestone)?

PR #1511 (2022-08-15): Make leap_day=True default for PSM3 (deprecation)

 - [x] Closes #1481 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~~
 - ~~[ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to  [... cut]
PR #1991 (2024-06-02): Make leap_day=True default for PSM3 (new)

 - [x] Closes #1481
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~~[ ] Tests added~~
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issu [... cut]
closed 2022-08-15 by AdamRJensen