pvlib/pvlib-python #1487: Accessing spectral transmission

lalligagger, 2022-07-03:
Hi, I'm wondering if it's possible to access the spectral atmospheric transmission when modeling spectral irradiance.
I'd like to propagate the calculated irradiance on the ground back up to top of atmosphere. Thanks! Following this tutorial: https://pvlib-python.readthedocs.io/en/stable/gallery/spectrum/plot_spectrl2_fig51A.html#id1

lalligagger, 2022-07-03:
I'm guessing this is what I want to pull out?
https://github.com/pvlib/pvlib-python/blob/c8b04a569e434bbb0dd7fc286bb6872bf3131f97/pvlib/spectrum/spectrl2.py#L168

Should it be same length as 'wavelength' in output dict?

mikofski, 2022-07-03:
I think for this you would need a more detailed radiative transfer model (RTM) like http://libradtran.org/doku.php. I believe there’s an old issue somewhere in pvlib related to this topic. Wikipedia has a long list of RTM’s:
https://en.wikipedia.org/wiki/Atmospheric_radiative_transfer_codes

You might need atmospheric conditions from the ground to the top of the atmosphere like the NASA 1976 standard atmosphere
https://ntrs.nasa.gov/citations/19770009539

lalligagger, 2022-07-03:
Ok thank you. I have some reference atmospheres I could use but it seemed desirable to have both directions match.
It's possible I'm not understanding the Bird model right as I've just started looking into it. Thanks for your reply.

closed 2022-07-03 by lalligagger