pvlib/pvlib-python #1488: Irradiance Model for solar plants in slope terrain

JoseVinicius1998, 2022-07-06:
Second, my studies about the PVLib package don't have an irradiance model for solar plants in sloped terrains. I see; there is the tracking algorithm for slope terrain, but not for the irradiance model. 
The features would be essential because there isn't this model in any other package, Python, and this implementation certainly will help some projects and academic studies. 
At this moment, I am studying the development of an algorithm to improve the harvesting irradiance, considering bad weather in a solar plant installed in sloped terrain with bifacial panels. It would be perfect for this model to finish my studies.

mikofski, 2022-07-06:
@kanderso-nrel I believe we considered applying the same rotation matrices from “slope-aware backtracking” to the ground plane in the “infinite sheds” model (rotating the solar vector to find the equivalent horizontal basis) but we decided to delay that until a later PR. If completed now, that might almost accommodate the POA calculation of panels on arbitrary slope except that the panels would still be constrained to a specific orientation wrt the ground, so *not* completely arbitrary. EG if the module surfaces    have different `(az, ze)` than ground not sure if we would have that covered, maybe?

kanderso-nrel, 2022-07-06:
@mikofski I think there are some secondary things to consider in addition to rotating the solar vector and module orientations into the ground's reference frame.  I haven't thought it through completely, but at least GCR is affected by frame rotations, and `poa_ground_diffuse` wouldn't be based on GHI anymore, I think you'd have to transpose DNI & DHI onto the tilted ground plane instead.  There's also the issue that the sky dome isn't the full 180 degrees anymore since it has to cede some to the terrain ("diffuse horizon shading"?).  Maybe it would be straightforward to accommodate all of that, but I think it would take some thought to make sure nothing is being overlooked.  

@JoseVinicius1998, for estimating insolation gains from improved tracking, you might be interested in the references cited in this small conference paper: https://www.nrel.gov/docs/fy22osti/82811.pdf

kanderso-nrel, 2022-07-06:
Ah I was just thinking about how one might be able to "trick" the existing `pvlib.bifacial.infinite_sheds` code (which assumes flat terrain in several places) into modeling a sloped system in a conceptually consistent way.  Wasn't thinking at all about how to cleanly integrate something like that into the pvlib classes.

JoseVinicius1998, 2022-07-07:
@kanderso-nrel, this reference is exactly what I have studied; excellent paper. I have also used the `pvlib.bifacial.infinite_sheds` function in my simulations, and I believe the only difference would be my wish to consider a sloped terrain.

I let my thanks. Congratulations sirs. The PVLib package is a **fantastic tool** for solar plant simulations.

closed 2022-07-11 by JoseVinicius1998

mikofski, 2023-01-19:
@kanderso-nrel it would be nice to revisit this goal of expanding infinite sheds or some 2D model to arbitrary mono-slopes. Does anyone have a reference we can use to go forward. I think the simple rotation matrix approach (like what we used for slope aware backtracking) can work here with some caveats on slope.
1. we can back out the GCR for the rotated reference frame
2. GHI may not differ much if the slopes are modest, or we scale the GHI by the ratio of the horizontal distance between rows versus the distance along the slope, EG: `GHI_slope = GHI*horiz_r2r_dist / sloped_r2r_dist = GHI*cos(slope_beta)` or ~GHI for slopes <3-arcdegrees b/c [`cos(3[deg]) = 0.99863](https://www.google.com/search?q=cos%283+degrees%29)
3. ignore changes in the sky dome, in the rotated reference frame the part it cedes to the ground is recovered by the slope in an infinite mono-plane

I think it's worth at least a try for slopes <10-deg (cos(10[deg]) = 0.985)

kanderso-nrel, 2023-01-19:
I don't have a reference.  I have a nebulous goal of writing a paper extending previous view factor models to accommodate these more complex cases, but no concrete timeline to get it done.

mikofski, 2023-01-19:
Maybe someone from SolarFarmer  team could collaborate and/ or support? Do you know Javier Lorente? (@jalpelo)

mikofski, 2023-03-23:
@pvlib/pvlib-core Coming back to this - at the very least, ignoring horizon, we could handle arbitrary azimuth easily enough by rotating the reference of the solar vector around the z-axis to line up with the system so that it appeared to be N*-S* aligned, this solves any horizon issues for now right?

I am looking at a dual-tilt case right now with modules facing slightly SE and NW, and I have to model it multiple times because I can't use infinite sheds to get the bifacial. That's crazy. I should be able to just enter the system azimuth and get the bifacial performance in one go. I'll try to put together a PR. Are we open to that?