pvlib/pvlib-python #1496: Sample consistent currents when create IV curves with pvlib.pvsystem.singlediode

jpplanted, 2022-07-21:
New to solar, so please bear with me.

**Is your feature request related to a problem? Please describe.**
It's much easier and intuitive to create string IV curves when the currents for the IV points are already aligned.

Currently, pvlib.pvsystem.singlediode creates (when using "newton") a log-spaced DIODE voltage sample set and then samples at those voltages. The returned voltages and currents are not guaranteed to be aligned. It would be nice to have an easy/intuitive way to get either current-aligned or voltage-aligned iv-curve samples.

bishop88 is a close-form i-v solution as a function of  diode voltage, so I don't know what is possible, other than solving bishop88_i_from_v or bishop88_v_from_i for the returned sample points.

**Describe the solution you'd like**
bishop88_i_from_v or bishop88_v_from_i called after sampling iv curves in pvlib.pvsystem.singlediode.

**Describe alternatives you've considered**
I'm pretty new here, and I don't know of many. I think bishop88_i_from_v and bishop88_v_from_i are the only options?

**Additional context**
I'm trying to do relatively high-accuracy MPPT simulation by building representative IV curves of modules -> strings -> MPPT. For instance, I want to try to increase the voltage when MPPT power is above its limit. Or current limit in the rare chance that this power-adjusted operating point is still above the max mppt/dc port continuous current.

 Please let me know if I'm missing something very obvious!

cwhanse, 2022-07-22:
HI @jpplanted 

> The returned voltages and currents are not guaranteed to be aligned.

I don't understand this statement - could you elaborate? The elements of the voltage and current vectors returned by `singlediode` are pairs which solve the single diode equation. Do you mean that the voltage (or current) values are not common for two different modules in a string? If that's correct, are you wanting to specify e.g. current values and then get the corresponding voltage?

If I've got that right, extending `pvlib.pvystem.singlediode` to accept (optional) user-specified voltage or current has been of interest for a while.

In the meantime, you can use `bishop88_i_from_v` etc. (or the private functions `singlediode._lambertw_i_from_v` etc.) to do what you describe.

By "high accuracy" what precision are you looking for? With default settings, the pvlib methods should provide 8-10 digits of precision.

cwhanse, 2024-11-12:
Closing because the capability desired by the OP is provided in `pvlib.pvsystem.i_from_v` and `pvlib.pvsystem.v_from_i`, and we have removed the `ivpoints` kwarg from `pvlib.pvsystem.singlediode`.

closed 2024-11-12 by cwhanse