pvlib/pvlib-python #1506: v0.9.2 release plan

cwhanse, 2022-07-26:
We plan to release v0.9.2 in the next 3 weeks. @pvlib/pvlib-maintainer please add to the list of PRs to merge and issues to close for v0.9.02.

- [ ] #1468 
- [ ] #1478

kanderso-nrel, 2022-07-29:
Can we drop python 3.6 support in this release?  GitHub Actions is dropping it in MacOS pretty soon, it's way past the NEP29 and Python end-of-life windows, and pvanalytics and rdtools have already moved to >=3.7.

mikofski, 2022-07-29:
I’m ok to drop 3.6 ✅

adriesse, 2022-08-16:
I guess I can't edit this myself, but I would like the spectral mismatch PR to go into this release because I would like to use the new functions in an example notebook to accompany a new open data set.

wholmgren, 2022-08-16:
Releasing on pypi and conda-forge is a couple minutes of work. We should just do it this week whatever we have. It would be really nice to speed up the release cadence and make it less of a big deal.

kanderso-nrel, 2022-08-19:
Unless someone already has a branch started, I will submit a PR shortly for final 0.9.2 cleanup.  Doesn't mean we can't do some last minute feature merges if the reviewing capacity is available (unfortunately I myself am unlikely to review anything else before the release); I'm just getting antsy about getting this out the door before PVPMC.

wholmgren, 2022-08-19:
@kanderso-nrel go for it. I'm +1 on #1478 if you think it's ready.

cwhanse, 2022-08-19:
I edited the checklist to defer PRs that aren't quite ready (either need reviews or author responses). The only one to decide is #1478

adriesse, 2022-08-19:
As I explained above, it would be nice to have the mismatch feature released fairly soon.  If it is not in this release, how much time do we think might elapse before the next release?

wholmgren, 2022-08-19:
@adriesse if a pre-release is acceptable then we can make that within minutes of merging the PR. ``git fetch upstream; git rebase upstream/master; git tag v0.9.3-alpha.1; git push upstream v0.9.3-alpha.1;`` wait a few minutes and then ``pip install pvlib --pre``. In principle we can make that a full release - the only complication is the build for the pvlib channel is manual and can be tedious (see the release notes wiki). Finishing the what's new is tedious when the release grows to the current size, but easy if there's only a couple of items. If you want to take on the what's new prep then it would help get the release out the door faster. So, somewhere between minutes, days, and 1-3 months depending on needs/desires/contributions.

adriesse, 2022-08-19:
I don't think it's so critical that it warrants exceptional actions, but perhaps we could consider 9.3 in 2-3 weeks?  My temperature PR should be done then too.

PR #1534 (2022-08-19): Finalize 0.9.2
 - [x] Closes #1506
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` ` [... cut]
closed 2022-08-19 by wholmgren