pvlib/pvlib-python #1508: update `benchmarks/tracking.py` for `SingleAxisTracker` deprecation

wholmgren, 2022-07-29:
**Describe the bug**
`SingleAxisTracker` is deprecated, but the `tracking.py` benchmark still uses it:

https://github.com/pvlib/pvlib-python/blob/3f397ed44075b01f4ba535750356859195a51b2d/benchmarks/benchmarks/tracking.py#L20

**To Reproduce**
run asv tests. or look at the asv quick CI output

**Expected behavior**
no deprecation warnings in the benchmark test suite

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``: 0.9. https://github.com/pvlib/pvlib-python/pull/1507

Naman-Priyadarshi, 2022-08-05:
I wanted to work on the issue! Should i just remove the `time_tracker_singleaxis` function from the benchmark file? 

Also when running `asv dev` I noticed that some of the benchmarking functions were failing, should i open a new issue regarding the same?

wholmgren, 2022-08-12:
Thanks @Naman-Priyadarshi, that would be great. Yes to both of your questions.

PR #1535 (2022-11-11): Remove benchmarks for deprecated SingleAxisTracker 

 - [x] Closes #1508
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] New  [... cut]
closed 2022-11-11 by kanderso-nrel