pvlib/pvlib-python #1515: Update reference in pvlib.iotools.read_cams

Kaesekopf, 2022-08-03:
**Problem:

In the near future the documentation of the CAMS radiation service will be centralised under https://atmosphere.copernicus.eu/solar-radiation, and replace the two references that are mentioned in the function read_cams.

**Proposed solution:

Short term:
Set "https://atmosphere.copernicus.eu/solar-radiation" as first reference in the reference list. 

Long term:
Once all documentation is moved to the new web page remove the two existing references.

AdamRJensen, 2022-08-03:
@Kaesekopf Thank you very much for this feedback!

Do you know if the link used in the documentation of the `get_cams` function will also be removed? https://www.soda-pro.com/help/cams-services/cams-mcclear-service/automatic-access

Will additional documentation that is more relevant than (https://atmosphere.copernicus.eu/solar-radiation) be added? Otherwise, we can just go ahead now and remove the two existing references and replace them with the single reference you mentioned.

Also, can you confirm that the link covers both CAMS Radiation and CAMS McClear?

Kaesekopf, 2022-08-03:
Hallo Adam,

Thanks for your (extremely) fast response. To answer your questions: yes in the future all CAMS radiation Service documentation will be centralised on (https://atmosphere.copernicus.eu/solar-radiation), so also the ones referenced in get_cams. But not all information in the old references are transferred yet, so for the moment it is wiser to keep the old references as well.

Sunny greetings,

Jethro

---------------------
Deutsches Zentrum für Luft- und Raumfahrt e.V. (DLR)
German Aerospace Center

Institut für Vernetzte Energiesysteme
Institute of Networked Energy Systems
Energiesystemanalyse | Carl-von-Ossietzky-Str. 15 | 26129 Oldenburg | Germany
Visiting address: DLR-Institut für Vernetzte Energiesysteme | Markt 6 | 26121 Oldenburg | Germany

Jethro Betcke M. | Scientist
Telephone +49 441 99906-293  | Telefax +49 441 99906-109 | ***@***.******@***.***>
DLR.de<https://owa.dlr.de/owa/redir.aspx?C=c_MWIozn853hOVNmaoxq7dr07VZXPZv0lVZrSGnNw4M7xIKn1G7VCA..&URL=http%3a%2f%2fwww.dlr.de%2f>/ve

Kaesekopf, 2022-08-03:
Hallo Adam,

I did not scroll al the way to the bottom of your mail. So to answer your last question: yes the new webpage will include information on both services. But as you can see the new webpage is still quite rudimentary, so the old references are still needed for the moment.

Sunny greetings,

Jethro

---------------------
Deutsches Zentrum für Luft- und Raumfahrt e.V. (DLR)
German Aerospace Center

Institut für Vernetzte Energiesysteme
Institute of Networked Energy Systems
Energiesystemanalyse | Carl-von-Ossietzky-Str. 15 | 26129 Oldenburg | Germany
Visiting address: DLR-Institut für Vernetzte Energiesysteme | Markt 6 | 26121 Oldenburg | Germany

Jethro Betcke M. | Scientist
Telephone +49 441 99906-293  | Telefax +49 441 99906-109 | ***@***.******@***.***>
DLR.de<https://owa.dlr.de/owa/redir.aspx?C=c_MWIozn853hOVNmaoxq7dr07VZXPZv0lVZrSGnNw4M7xIKn1G7VCA..&URL=http%3a%2f%2fwww.dlr.de%2f>/ve

AdamRJensen, 2022-08-03:
I'm inclined to wait with changing the references until the new documentation becomes available since the old links still work. This avoids making two pull requests and saves a bit of work. However, if you think it's important and useful to add the other link now, I'd also be happy to do that.

Kaesekopf, 2022-08-03:
OK, that sounds reasonable. I propose that I will reply to this thread again when the transfer is complete.

Kaesekopf, 2022-08-04:
Hallo Adam,

We had a slight internal misunderstanding within our project. The new CAMS Radiation Service documentation is to be found at:

https://confluence.ecmwf.int/display/CKB/CAMS+solar+radiation+time-series%3A+data+documentation

All the documentation for CAMS Radiation and CAMS McClear is already at this new website. So both the old sodapro references should be replaced by this single new one. 

Thanks,

Jethro

AdamRJensen, 2022-08-04:
This is fine with me. However, I think we need to keep this [link](https://www.soda-pro.com/help/cams-services/cams-mcclear-service/automatic-access) in the documentation of the `get_cams` function as it describes the SodaPro API being used.

I assume that SodaPro will continue to serve data and that CAMS Radiation will not only be available from the [Atmosphere Data Store](https://ads.atmosphere.copernicus.eu/cdsapp#!/dataset/cams-solar-radiation-timeseries?tab=overview)?

AdamRJensen, 2022-08-08:
@Kaesekopf I didn't tag you in my response so perhaps you didn't get notified.

PR #1529 (2022-08-18): Update cams links

 - [x] Closes #1515
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 ~~- [ ] Tests added~~
 ~~- [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue [... cut]
closed 2022-08-18 by AdamRJensen