pvlib/pvlib-python #1522: The function 'get_processed_data' can't get data at the website

q33420540, 2022-08-11:
The function 'get_processed_data' can't get data at the website

The server will return:
Server Error (400: NETCDF3 format not supported for ANY_POINT feature type.)

**************************
Error accessing https://thredds.ucar.edu/thredds/ncss/grid/grib/NCEP/GFS/Global_0p5deg/Best?var=Downward_Short-Wave_Radiation_Flux_surface_Mixed_intervals_Average&var=u-component_of_wind_isobaric&var=High_cloud_cover_high_cloud_Mixed_intervals_Average&var=Total_cloud_cover_boundary_layer_cloud_Mixed_intervals_Average&var=v-component_of_wind_isobaric&var=Low_cloud_cover_low_cloud_Mixed_intervals_Average&var=Wind_speed_gust_surface&var=Total_cloud_cover_convective_cloud&var=Total_cloud_cover_entire_atmosphere_Mixed_intervals_Average&var=Temperature_surface&var=Medium_cloud_cover_middle_cloud_Mixed_intervals_Average&time_start=2022-08-11T07%3A00%3A00%2B00%3A00&time_end=2022-08-14T07%3A00%3A00%2B00%3A00&longitude=-110.9&latitude=32.2&vertCoord=100000&accept=netcdf

avarf, 2022-08-11:
I am facing this issue too.

cwhanse, 2022-08-11:
My first guesses are an issue with the server, or a change in server API. Please note that `pvlib/forecast.py` is deprecated and planned for removal, due to not having a maintainer for that capability. See notes [here](https://pvlib-python.readthedocs.io/en/stable/user_guide/forecasts.html).

closed 2022-08-12 by wholmgren

avarf, 2022-08-23:
I just started to work on this again and maybe I am missing something but there might be an easy fix for this. Everything is working (creating the query, getting the data, etc) and the only thing that fails is `_netcdf2pandas`.

To solve that I am using the below code which is not sensitive regarding data type, dimension, etc:
```python
import xarray as xr

dataset = xr.open_dataset(xr.backends.NetCDF4DataStore(data))
df = dataset.to_dataframe().reset_index()
```

As I said I just started to look into this and my goal is to create my own module to work with different weather feeds but I think this might solve this error.

josegardo, 2022-08-25:
Hi @avarf 

Can you explain where to put this code?

This is a simple test script for GFS forecasting and it is currently failing.

`import pandas as pd
import matplotlib.pyplot as plt

import datetime

# import pvlib forecast models
from pvlib.forecast import GFS

# specify location (Tucson, AZ)
latitude, longitude, tz = 36, -6, 'UTC'

# specify time range.
start = pd.Timestamp(datetime.date.today(), tz=tz)

end = start + pd.Timedelta(days=7)

# GFS model, defaults to 0.5 degree resolution
# 0.25 deg available
model = GFS()

# retrieve data. returns pandas.DataFrame object
raw_data = model.get_data(latitude, longitude, start, end)

print(raw_data.head())`

Thank you!

avarf, 2022-08-25:
Since the GFS of the PVlibe is not supported anymore I followed the suggested solution and used siphon. To use siphon you need to create the queries yourself and because of some changes on the server side you can't use the dictionary that PVlibe is using and you need to break the query into 4 sub-queries (look at the other issue that I opened at unidata repo).

The example that unidata provides works for me and I copy it for one of the queries here:
```python
import requests
import pandas as pd
import io
from siphon.catalog import TDSCatalog
from siphon.ncss import NCSS
from datetime import datetime, timedelta
from netCDF4 import num2date
from pvlib.location import Location
import xarray as xr

top_cat = TDSCatalog('http://thredds.ucar.edu/thredds/catalog.xml')
models_cat = top_cat.catalog_refs[0].follow()
gfs_cat = models_cat.catalog_refs['GFS Quarter Degree Forecast'].follow()
ds = gfs_cat.latest
ncss = ds.subset()

query1 = ncss.query()
query1.lonlat_point(lon=-105, lat=40) # set coordinates of point of interest.
now = datetime.utcnow() # get current time
query1.time_range(now, now + timedelta(days=1)) # create time range of 24 hours

query1.variables('Total_cloud_cover_entire_atmosphere_Mixed_intervals_Average', 'Medium_cloud_cover_middle_cloud_Mixed_intervals_Average', 'High_cloud_cover_high_cloud_Mixed_intervals_Average', 'Total_cloud_cover_boundary_layer_cloud_Mixed_intervals_Average','Low_cloud_cover_low_cloud_Mixed_intervals_Average','Downward_Short-Wave_Radiation_Flux_surface_Mixed_intervals_Average')

query1.accept('netcdf4') # return data as a netCDF4 object
point_data1 = ncss.get_data(query1)

raw_data1 = ncss.get_data(query1)
dataset1 = xr.open_dataset(xr.backends.NetCDF4DataStore(raw_data1))
df1 = dataset1.to_dataframe().reset_index()
```
* Probably there are some imports that are not needed but I just copy paste :)

After you get all the queries you need to join them and later process the data to be in the required format for PV (have GHI, DHI, DNI, etc) and this is what I am currently working on.

josegardo, 2022-08-25:
Thk @avarf 

This code has been very helpful