pvlib/pvlib-python #1526: Benchmarking fucntions failing

Naman-Priyadarshi, 2022-08-14:
**Describe the bug**
`solarposition.SolarPosition.time_pyephem` , `solarPosition.time_sun_rise_set_transit_ephem` and `SolarPositionCalcTime.time_calc_time` benchmarking functions are failing, the fucntions are a part of solarposition.py

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'benchmarks'
2. Open in terminal
3. Run asv using 'asv dev'
4. See error

**Expected behavior**
The benchmarks should run without any errors.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Versions:**
 - python: 3.8.10
 - asv: 0.5.2

jules-ch, 2022-08-16:
you're missing pyephem to run the full benchmark.

`pip install ephem`

kanderso-nrel, 2022-08-16:
Yeah this seems like a local problem; the nightly job seems to be running those benchmarks without trouble: https://pvlib-benchmarker.github.io/pvlib-benchmarks/#solarposition.SolarPosition.time_pyephem?x-axis-scale=date

@Naman-Priyadarshi I think you can specify `--show-stderr` to get additional info beyond just `failed`.

Naman-Priyadarshi, 2022-08-17:
yes, the problem was because numba and pyephem was missing on my system, thank you @kanderso-nrel @jules-ch

closed 2022-08-17 by Naman-Priyadarshi