pvlib/pvlib-python #1541: Reduced order electrical mismatch model

kanderso-nrel, 2022-08-29:
**Is your feature request related to a problem? Please describe.**
Electrical mismatch loss due to cell-to-cell irradiance nonuniformity is relevant for bifacial system performance, but pvlib does not offer any tools to model this.  PVMismatch can, but maybe I don't want to learn a new tool, maybe I don't have 2-diode model parameters, maybe I don't want another dependency, maybe I just want a quick mismatch estimate without the computational burden of a full I-V circuit solver...

**Describe the solution you'd like**
One of the many feature requests suggested at PVPMC (thanks @shirubana!) is an implementation of the reduced order electrical mismatch model described in [Deline, Pelaez et al 2020, "Estimating and parameterizing mismatch power loss in bifacial photovoltaic systems"](https://doi.org/10.1002/pip.3259).  

The model itself is straightforward: given irradiance at several locations across a module, summarize the irradiance variation using mean absolute difference and feed that into a polynomial with known coefficients.  What might be more difficult is deciding what the structure of the inputs should be, as we don't currently have a data structure convention for representing multiple irradiance values for a single module.  It must be at least 2D (time, location in module), but maybe 3D would be better to allow vectorization across modules.

**Describe alternatives you've considered**
Of course a full I-V curve circuit simulator is still desirable in the long run, but this model is a lot simpler both to implement and to use.  No reason not to have both IMHO.

**Additional context**

- `pvfactors` will give you localized/partitioned irradiance going up the row height if you ask it nicely; perhaps we should add a gallery example demonstrating how to do that.  I don't think the infinite sheds code can return irradiance for small parts of the module, can it?
- `bifacial_radiance` already has a BSD-3 implementation we could use as a starting point:
https://github.com/NREL/bifacial_radiance/blob/5f9de36861ec9bcffe169c9a4371806860c0e224/bifacial_radiance/mismatch.py#L166-L220

shirubana, 2022-08-30:
Which module do you envision this function to live?

cwhanse, 2022-08-30:
pvsystem.py for this simple model, I think. Code for detailed mismatch calculations would go into singlediode or ivtools.

PR #2046 (2024-07-01): NREL non-uniform irradiance mismatch loss for bifacial modules
 - [x] Closes #1541
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to [... cut]
closed 2024-07-01 by kandersolar