pvlib/pvlib-python #1543: add snyk open source advisor badge

mikofski, 2022-09-06:
**Is your feature request related to a problem? Please describe.**
Do people wonder if pvlib and open source are safe? Snyk is one of several online solutions that offer security analysis. According to them pvlib is looking pretty good

[image]

**Describe the solution you'd like**
Add Snyk OS advisor in the badge matrix on the readme.

[image]

**Describe alternatives you've considered**
* https://deps.dev/pypi/pvlib is an OpenSSF scorecard. pvlib scores 6/10
* IDK?

**Additional context**
This is how you add the markdown and it should take you to the snyk OS advisor page:

[image]

kanderso-nrel, 2023-01-24:
I'm not sure I put much stock in an endorsement from snyk and I'm not sure we would be doing readers a service by directing them to snyk's page on pvlib.  It's way at the bottom of the page, but this part gives me pause:
> **Is pvlib safe to use?**
> The python package pvlib was scanned for known vulnerabilities and missing license, and no issues were found. Thus the package was deemed as safe to use. See the full [health analysis review](https://snyk.io/advisor/python/pvlib#security).

I don't think this website has sufficient justification to claim pvlib is "safe to use" (not that I disagree with the conclusion of course, just its reasoning).  One could create an MIT-licensed `maliciouslib` that contains only `import os; os.system("sudo rm -rf /")` and I think this website would claim it is safe to use because it has a license and there are no known vulnerabilities in its zero dependencies.

mikofski, 2023-01-25:
but people (like my employer) are paying money to Snyk for these assessments?

As an alternate maybe we can link to the Black Duck security assessment on OpenHub.net - it also approves of pvlib threadwise, and I think its testing is a little more comprehensive?
https://openhub.net/p/pvlib-python

kanderso-nrel, 2023-01-25:
Maybe I am unnecessarily critical... far be it from me to gainsay DNV on this :)  If you see value in this kind of analysis/report, don't let my curmudgeonry stand in the way!

AdamRJensen, 2023-01-25:
Let's just add it. I vote calling the caption something along the lines of "security" as this gives some indication of what it is - I imagine a lot of people might not be familiar with Snyk.

mikofski, 2023-01-26:
Kevin, I value your curmudgeonly stand more than most, and from what I could tell, the Snyk Advisor does seem a bit like a glorified advertisement for their paid services. Big companies paying for Snyk are probably getting a lot more than the free "Advisor" service, and there's also a climate of security concern as big business transitions to digital. 

I think there could be possible blowback from showing this badge, having folks see it and believe pvlib is immune from all malware, and then NumPy getting hacked and causing our downstream users to get affected as well. I don't want that responsibility!

Maybe we look and see are there any other sites that are using these Snyk badges? Is there a disclaimer we should add somewhere? Maybe it is time for us to add a security statement?

|proj| has snyk |
|---|---|
|numpy| no |
|scipy|no|
|MPL|no|
|pandas|no|
|jupyter notebook|no|

In fact I couldn't find any packages in the Python stack that DO use the Snyk Advisor badge. I've posted a question on NF slack channel, but I'm steering away from adding this.

cwhanse, 2023-01-26:
If none of those projects are using snyk, my vote is no.

adriesse, 2023-01-26:
My first thought was: why only 90/100?  Where is pvlib unhealthy?  I didn't have that question before I saw the badge.

closed 2023-01-27 by mikofski