pvlib/pvlib-python #1545: v0.9.3 release plan

adriesse, 2022-09-06:
Do we need one of these yet?

kanderso-nrel, 2022-09-08:
@adriesse do you have a specific deadline for this release?  I suspect that we'll drag our feet here without one. 

Seems like the main things we want to get into this release are @adriesse's two PRs:
- [ ] #1463
- [ ] #1524

Others that I'd like to see merged but are less time sensitive:
- [ ] #1468 -- This is very close, just one open [question about units](https://github.com/pvlib/pvlib-python/pull/1468#discussion_r924785711)
- [ ] #1542 

I'd be okay with bumping the other items on the 0.9.3 [milestone](https://github.com/pvlib/pvlib-python/milestone/37) to the next release.

adriesse, 2022-09-08:
I think that Sept. 15 would make a suitable target date.  I don't want too push too hard because the reviews and discussions are important.  But if there is broad agreement that these PR's are ready then it could be earlier.

kanderso-nrel, 2022-09-14:
@pvlib/pvlib-maintainer, is there anything else people would like to see included in 0.9.3?  Anticipating an answer of "no", I have moved all the 0.9.3 milestone items to [0.9.4](https://github.com/pvlib/pvlib-python/milestone/38).  I'll open a finalization PR shortly.

cwhanse, 2022-09-14:
I can do it.

mikofski, 2022-09-15:
Thanks for driving it home. I think the release is ready?

PR #1552 (2022-09-15): Finalize 0.9.3
 - [x] Closes #1545 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num`  [... cut]
closed 2022-09-15 by kanderso-nrel